cjcliffe/CubicSDR #31: ../src/unix/glx11.cpp(86): assert "xid" failed in SetCurrent(): window must be shown

pwarren, 2015-01-19:
I'm running Debian 8 with liquidSDR and rtl-sdr built from source, others from system packages. It builds fine, but at runtime I get an error :(  

I don't have my rtl-sdr dongle with me, so that might be an issue, but will test when I get home!

Backtrace follows below, as well as build info.

```
pwarren@marly:~/Projects/CubicSDR/build(master)$ ./CubicSDR 
RTL Devices: 0
SDR thread initializing..
SDR post-processing thread started..
../src/unix/glx11.cpp(86): assert "xid" failed in SetCurrent(): window must be shown
RTL Devices: 0
No devices found.. SDR Thread exiting..
Loaded font 'Bitstream Vera Sans Mono' from 'vera_sans_mono12.fnt', parsed 167 characters.
Loaded font 'Bitstream Vera Sans Mono' from 'vera_sans_mono16.fnt', parsed 167 characters.
Loaded font 'Bitstream Vera Sans Mono' from 'vera_sans_mono18.fnt', parsed 167 characters.
Loaded font 'Bitstream Vera Sans Mono' from 'vera_sans_mono24.fnt', parsed 167 characters.
Loaded font 'Bitstream Vera Sans Mono' from 'vera_sans_mono32.fnt', parsed 167 characters.
Loaded font 'Bitstream Vera Sans Mono' from 'vera_sans_mono48.fnt', parsed 167 characters.
../src/unix/glx11.cpp(86): assert "xid" failed in SetCurrent(): window must be shown
../src/unix/glx11.cpp(86): assert "xid" failed in SetCurrent(): window must be shown
../src/unix/glx11.cpp(86): assert "xid" failed in SetCurrent(): window must be shown
../src/unix/glx11.cpp(86): assert "xid" failed in SetCurrent(): window must be shown
../src/unix/glx11.cpp(86): assert "xid" failed in SetCurrent(): window must be shown
../src/unix/glx11.cpp(86): assert "xid" failed in SetCurrent(): window must be shown
../src/unix/glx11.cpp(86): assert "xid" failed in SetCurrent(): window must be shown
../src/unix/glx11.cpp(86): assert "xid" failed in SetCurrent(): window must be shown
../src/unix/glx11.cpp(86): assert "xid" failed in SetCurrent(): window must be shown
../src/unix/glx11.cpp(86): assert "xid" failed in SetCurrent(): window must be shown
../src/unix/glx11.cpp(86): assert "xid" failed in SetCurrent(): window must be shown
../src/unix/glx11.cpp(86): assert "xid" failed in SetCurrent(): window must be shown
../src/unix/glx11.cpp(86): assert "xid" failed in SetCurrent(): window must be shown
../src/unix/glx11.cpp(86): assert "xid" failed in SetCurrent(): window must be shown
../src/unix/glx11.cpp(86): assert "xid" failed in SetCurrent(): window must be shown
../src/unix/glx11.cpp(86): assert "xid" failed in SetCurrent(): window must be shown
../src/unix/glx11.cpp(86): assert "xid" failed in SetCurrent(): window must be shown
../src/unix/glx11.cpp(86): assert "xid" failed in SetCurrent(): window must be shown

RtApiDummy: This class provides no functionality.

Terminating SDR thread..
Terminating SDR post-processing thread..
SDR post-processing thread done.
```

ASSERT INFO:
../src/unix/glx11.cpp(86): assert "xid" failed in SetCurrent(): window must be shown

BACKTRACE:
[1] wxGLContext::SetCurrent(wxGLCanvas const&) const
[2] PrimaryGLContext::PrimaryGLContext(wxGLCanvas*, wxGLContext*) /home/pwarren/Projects/CubicSDR/src/visual/PrimaryGLContext.cpp:58
[3] CubicSDR::GetContext(wxGLCanvas*) /home/pwarren/Projects/CubicSDR/src/CubicSDR.cpp:121
[4] ModeSelectorCanvas::ModeSelectorCanvas(wxWindow*, int*) /home/pwarren/Projects/CubicSDR/src/visual/ModeSelectorCanvas.cpp:30
[5] AppFrame::AppFrame() /home/pwarren/Projects/CubicSDR/src/AppFrame.cpp:45
[6] CubicSDR::OnInit() /home/pwarren/Projects/CubicSDR/src/CubicSDR.cpp:71
[7] wxAppConsoleBase::CallOnInit() /usr/include/wx-3.0/wx/app.h:93
[8] wxEntry(int&, wchar_t**)
[9] main /home/pwarren/Projects/CubicSDR/src/CubicSDR.cpp:16
[10] __libc_start_main
[11] _start

```
cmake ../ -DCMAKE_BUILD_TYPE=DEBUG
-- The C compiler identification is GNU 4.9.1
-- The CXX compiler identification is GNU 4.9.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/x86_64-linux-gnu/libX11.so
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Found wxWidgets: TRUE  
-- Copying /home/pwarren/Projects/CubicSDR/font/*.fnt to directory /home/pwarren/Projects/CubicSDR/build
-- Configuring file vera_sans_mono32.fnt
-- Configuring file vera_sans_mono24.fnt
-- Configuring file vera_sans_mono48.fnt
-- Configuring file vera_sans_mono12.fnt
-- Configuring file vera_sans_mono18.fnt
-- Configuring file vera_sans_mono16.fnt
-- Copying /home/pwarren/Projects/CubicSDR/font/*.png to directory /home/pwarren/Projects/CubicSDR/build
-- Configuring file vera_sans_mono18_0.png
-- Configuring file vera_sans_mono32_0.png
-- Configuring file vera_sans_mono24_0.png
-- Configuring file vera_sans_mono48_0.png
-- Configuring file vera_sans_mono16_0.png
-- Configuring file vera_sans_mono12_0.png
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pwarren/Projects/CubicSDR/build
```
```
pwarren@marly:~/Projects/CubicSDR/build(master)$ make -j4
Scanning dependencies of target CubicSDR
[  5%] [  5%] [ 11%] [ 11%] Building CXX object CMakeFiles/CubicSDR.dir/src/AppFrame.cpp.o
Building CXX object CMakeFiles/CubicSDR.dir/src/sdr/SDRThread.cpp.o
Building CXX object CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o
Building CXX object CMakeFiles/CubicSDR.dir/src/sdr/SDRPostThread.cpp.o
[ 13%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorPreThread.cpp.o
[ 16%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorThread.cpp.o
[ 19%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorWorkerThread.cpp.o
[ 22%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorInstance.cpp.o
[ 25%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorMgr.cpp.o
[ 27%] Building CXX object CMakeFiles/CubicSDR.dir/src/audio/AudioThread.cpp.o
[ 30%] Building CXX object CMakeFiles/CubicSDR.dir/src/util/Gradient.cpp.o
[ 33%] Building CXX object CMakeFiles/CubicSDR.dir/src/util/Timer.cpp.o
[ 36%] Building CXX object CMakeFiles/CubicSDR.dir/src/util/MouseTracker.cpp.o
[ 38%] Building CXX object CMakeFiles/CubicSDR.dir/src/util/GLFont.cpp.o
[ 41%] Building CXX object CMakeFiles/CubicSDR.dir/src/util/DataTree.cpp.o
[ 44%] Building CXX object CMakeFiles/CubicSDR.dir/src/visual/ColorTheme.cpp.o
[ 47%] Building CXX object CMakeFiles/CubicSDR.dir/src/visual/PrimaryGLContext.cpp.o
[ 50%] Building CXX object CMakeFiles/CubicSDR.dir/src/visual/InteractiveCanvas.cpp.o
[ 52%] Building CXX object CMakeFiles/CubicSDR.dir/src/visual/MeterCanvas.cpp.o
[ 55%] Building CXX object CMakeFiles/CubicSDR.dir/src/visual/MeterContext.cpp.o
[ 58%] Building CXX object CMakeFiles/CubicSDR.dir/src/visual/TuningCanvas.cpp.o
[ 61%] Building CXX object CMakeFiles/CubicSDR.dir/src/visual/TuningContext.cpp.o
[ 63%] Building CXX object CMakeFiles/CubicSDR.dir/src/visual/ModeSelectorCanvas.cpp.o
[ 66%] Building CXX object CMakeFiles/CubicSDR.dir/src/visual/ModeSelectorContext.cpp.o
[ 69%] Building CXX object CMakeFiles/CubicSDR.dir/src/visual/ScopeCanvas.cpp.o
[ 72%] Building CXX object CMakeFiles/CubicSDR.dir/src/visual/ScopeContext.cpp.o
[ 75%] Building CXX object CMakeFiles/CubicSDR.dir/src/visual/SpectrumCanvas.cpp.o
[ 77%] Building CXX object CMakeFiles/CubicSDR.dir/src/visual/SpectrumContext.cpp.o
[ 80%] Building CXX object CMakeFiles/CubicSDR.dir/src/visual/WaterfallCanvas.cpp.o
[ 83%] Building CXX object CMakeFiles/CubicSDR.dir/src/visual/WaterfallContext.cpp.o
[ 86%] Building CXX object CMakeFiles/CubicSDR.dir/external/rtaudio/RtAudio.cpp.o
[ 88%] Building CXX object CMakeFiles/CubicSDR.dir/external/lodepng/lodepng.cpp.o
[ 91%] Building CXX object CMakeFiles/CubicSDR.dir/external/tinyxml/tinyxml.cpp.o
[ 94%] Building CXX object CMakeFiles/CubicSDR.dir/external/tinyxml/tinystr.cpp.o
[ 97%] Building CXX object CMakeFiles/CubicSDR.dir/external/tinyxml/tinyxmlparser.cpp.o
[100%] Building CXX object CMakeFiles/CubicSDR.dir/external/tinyxml/tinyxmlerror.cpp.o
Linking CXX executable CubicSDR
[100%] Built target CubicSDR
```

cjcliffe, 2015-01-20:
Interesting, it looks like it's initializing the GL context before it's ready?  Are you using wxWidgets 3.0.2?

cjcliffe, 2015-01-20:
Try the latest master, I've meged the linux branch fixes which included some issues with textures initializing before the canvas was visible.

pwarren, 2015-01-20:
Yeah, the installed wx widgets are 3.0.2,

Using the latest master with #32 I do get a bit further, it creates an audio output thread and it displays better, without missing textures, but I still get the error, which comes up in a dialogue box.

cjcliffe, 2015-01-20:
I'll try re-ordering the initialization and make sure everything is in view, I'll update this ticket with the branch information when ready

cjcliffe, 2015-01-21:
Please check out the issue33-linux-setcurrent branch cjcliffe/CubicSDR@4dad30e9bd0043b2d28404b263f80d954f5f24e1 and let me know if that fixes your issue.  There were definitely several instances where SetCurrent() was being called prior to the actual draw calls.

pwarren, 2015-01-21:
Yep, all fixed, thanks :)

I have a BladeRF, and I'll look at hacking in support for that over the next few weeks.

closed 2015-01-21 by pwarren