cjcliffe/CubicSDR #35: Build Notes

pwarren, 2015-01-21:
Not sure how/where you would want to stick this, but I created some build notes fro Debian8:

```
sudo apt-get install git build-essential librtlsdr-dev automake
libfftw3-dev cmake libgl1-mesa-dev libwxgtk3.0-dev libpulse-dev

git clone https://github.com/jgaeddert/liquid-dsp

cd liquid-dsp
./bootsrap.sh
./configure
make
sudo make install
sudo ldconfig
cd ..

git clone https://github.com/cjcliffe/CubicSDR.git
cd CubicSDR
mkdir build
cd build
cmake ../
make
./CubicSDR
```

cjcliffe, 2015-01-21:
Excellent thanks -- I'll be putting instructions in the Wiki but will leave this issue up as the go-to for now; I'm deploying a fresh Windows 8.1 VM at the moment so that I can prepare the build instructions, which will be a bit trickier..

yoesak, 2015-01-22:
@cjcliffe I still struggling with the errors on ThreadQueue.. could you point a clue? I use VS2012 (Native Compiler / 2012 CTP) but the error is same

cjcliffe, 2015-01-22:
@yoesak I wasn't able to make a windows build using MSVC and am as of yet unsure how to make it work; that said I didn't put in much effort to get it going as TDM64-GCC just worked for the most part.  I will provide instructions on how to use TDM64-GCC from a fresh 8.1 install and am working out the 32-bit instructions as well (it was originally working on 32bit, but an old version of MinGW32).

yoesak, 2015-01-22:
@cjcliffe thanks for pointing me a clue not to use MSVC :), I will wait for instruction then...

cjcliffe, 2015-01-22:
Mac Build Instructions (using MacPorts):
```
ccliffe$ sudo port install liquid-dsp fftw-3-single wxWidgets-3.0-libcxx
.. installing ..
ccliffe$ git clone https://github.com/cjcliffe/CubicSDR.git
.. cloning ..
ccliffe$ cd CubicSDR/cmake_build
ccliffe$ cmake ../ -DwxWidgets_CONFIG_EXECUTABLE=/opt/local/Library/Frameworks/wxWidgets.framework/Versions/wxWidgets/3.0-libcxx/bin/wx-config
... generating ...
-- Build files have been written to: .../CubicSDR/cmake_build
ccliffe$ make
.. compiling ..
ccliffe$ ./CubicSDR
```
If for some reason your wx-config is elsewhere (built manually or otherwise) you can find it via:
```
ccliffe$ locate wx-config
/opt/local/Library/Frameworks/wxWidgets.framework/Versions/wxWidgets/3.0-libcxx/bin/wx-config
```

I have a partially working BundleUtilities that produces an .app and .dmg package but haven't finished the resource location support and metadata yet.  I'll update once it's available.

cjcliffe, 2015-01-22:
@yoesak If you're eager and have 64-bit windows there is a test build up at: https://drive.google.com/file/d/0Bx8hq1ez_vazNEpabl9ucU0zMGs/view that was used for verifying the OpenGL update fixes yesterday.

yoesak, 2015-01-22:
@cjcliffe Thanks, I've just tried the app and it works with my NESDR-E4000, but I still wonder how to compile and build on Windows 8.1, hope you will post the instruction

cjcliffe, 2015-01-22:
"Truth" has seemingly had success with Bananna Pi via rtl-sdr.com and reports to do the following to get Liquid-DSP to compile:

```
$ vi makefile 
(ESC key)
:%s/mfloat-abi=soft/mfloat-abi=hard/g
:wq
$ 
```
Which basically means "edit makefile and replace all mfloat-abi=soft with mfloat-abi=hard" for the non-vi users.

bodrick, 2015-01-23:
So to chime in I was able to build on Windows 8.1 using mingw-w64 which is available at http://mingw-w64.sourceforge.net/.  I then had to build wxWidgets, when building wxWidgets, make sure to use -std=gnu++11 otherwise it won't compile.  Then once you've ran cmake and set up the proper libraries etc, you should be able to build.  I can try and create some proper steps when I get home later on today.

bodrick, 2015-01-23:
So I've also tried to compile using Visual Studio 2013, and after some tinkering got to the following issue and wonder if anyone can shed some light :
This line https://github.com/cjcliffe/CubicSDR/blob/master/src/sdr/SDRThread.cpp#L265 causes me issues, Visual C++ will complain that there is an ambiguous conversion.  

I'm not a C++ programmer so I'm not sure where to start, I am familiar and I program in other languages, namely C#, and VB.Net and frontend web stuff so I'm a little lost on how I would fix this.

cjcliffe, 2015-01-24:
I have visual studio ready to go on my main desktop, if you have any build steps up to that point I'll add them to the Windows build instructions I'm working on now.

For a quick fix you may try "if (iqDataOutQueue.load() != NULL)" as I think it's just complaining about comparing NULL with std::atomic and doesn't know if it should compare the container or it's pointer contents.

bodrick, 2015-01-24:
So good news, I managed to compile using Visual C++ 12.  I did have to make a couple of changes to the source to resolve some of the compile errors, and the suggestion about using .load worked great.
I had to generate lib files for the liquid-dsp, fftw and I compiled rtl-sdr, although I would imagine you can generate the lib file without having to recompile.
You need to run the VS2013 x64 Native Tools Command prompt, then find the def file for the dll you wish to generate the lib for then run : lib /def:libfftw3f-3.def, this will generate the lib file which you can then link against in VS.
I will try and do a better step by step hopefully tomorrow, and also set up a pull request with my changes.

yoesak, 2015-01-24:
Finally I can build with TDMGCC64 with the instruction above, And I will try using VS2012 with your clue @bodrick

cjcliffe, 2015-01-25:
Windows 8.1, 64-bit: *1st draft* -- comments and improvements welcome.

# Install MSYS:

Not sure if this is required, but would be needed if you want to build your own copy of some of the included DLL's and is how I set up my build environment.

Download MSYS from:
    http://sourceforge.net/projects/mingw/files/
For this guide I've used:
    http://downloads.sourceforge.net/project/mingw/Installer/mingw-get-setup.exe

- Run mingw-get-setup.
- Change the install path to "C:\MinGW\32" (this sets us up for optional 32-bit install later)
- Uncheck "Also install support for the GUI..." and click "Continue" and and finish the install.
- Edit "c:\mingw\32\var\lib\mingw-get\data\profile.xml" and replace contents with the following: 
```xml
<?xml version="1.0" encoding="UTF-8" standalone="yes"?>
<profile project="MinGW" application="mingw-get">
  <repository uri="http://prdownloads.sourceforge.net/mingw/%F.xml.lzma?download"/>
  <system-map id="default">
    <sysroot subsystem="mingw32" path="%R" />
    <sysroot subsystem="MSYS" path="%R/../msys" />
  </system-map>
</profile>
```
- Open a command prompt and run:
```
    c:\mingw\32\bin\mingw-get install msys-core msys-base msys-vim msys-wget msys-patch msys-flex msys-bison
```
- Create the file "c:\mingw\msys\etc\fstab" and put the following:
```
    c:/mingw/32   /mingw
```
- MSYS Batch file "msys.bat" will be in "C:\MinGW\msys", it's useful to make a shortcut somewhere accessible.

# Install TDM-GCC64:

Download TDM-GCC64 from:
    http://tdm-gcc.tdragon.net/ 
For this guide I've used:
    http://sourceforge.net/projects/tdm-gcc/files/TDM-GCC%20Installer/tdm64-gcc-4.9.2-3.exe/download

- Run the installer and install to "c:\mingw\64"
- Choose the "32 and 64-bit" install and continue to package selection and finish the installation

# Install wxWidgets:

Download wxWidgets from:
    http://wxwidgets.org/downloads/
For this guide I've used:
    http://downloads.sourceforge.net/project/wxwindows/3.0.2/wxWidgets-3.0.2.zip

- Extract .ZIP to a folder, i.e. "C:\Dev\wxWidgets-3.0.2".
- Open a command prompt and navigate to "C:\Dev\wxWidgets-3.0.2\build\msw".
- Run:
```
mingw32-make -f makefile.gcc BUILD=release UNICODE=1 USE_OPENGL=1 SHARED=0 CXXFLAGS="-fno-keep-inline-dllexport -std=gnu++11" 
```
- Wait for the build to finish; and if you built it in a Windows 8.1 VM on a late 2010 Macbook like I did, it might be time for a few beers.
- Onsch wxWIdges is build sbober up and continuw to nhxt step.,
    
# Install CMake:
    
Download CMake from:
    http://www.cmake.org/download/
For this guide I've used:
    http://www.cmake.org/files/v3.1/cmake-3.1.1-win32-x86.exe

Just install CMake with default or preferred options.

# Build CubicSDR:

- Clone or download zip from https://github.com/cjcliffe/CubicSDR/ to a folder of your choice.  I'll use C:\Dev\CubicSDR
- Run CMake GUI
- Choose C:\Dev\CubicSDR for source.
- Choose C:\Dev\CubicSDR\cmake_build for binaries.
- Click Configure.
- I usually choose Eclipse but for tutorial simplicity choose "MinGW Makefiles"
- Set CMAKE_BUILD_TYPE to "Release"
- Set wxWidgets_ROOT_DIR to "C:\Dev\wxWidgets-3.0.2"
- Configure again, all should be good, then Generate
- Open a command prompt and navigate to CubicSDR/cmake_build folder i.e. "cd C:\Dev\CubicSDR\cmake_build"
- Run "mingw32-make"
- CubicSDR.exe should be ready to run (assuming Zadig is already setup)

cjcliffe, 2015-01-28:
Windows 8.1, Visual Studio 64-bit: *1st draft* -- comments and improvements welcome.

# Install Visual Studio Express 2013

If you don't alreadu have Visual Studio you can install the Microsoft "Visual Studio Express 2013 for Windows Desktop" available from http://www.visualstudio.com/en-us/products/visual-studio-express-vs.aspx which was used for this guide.

Make sure you get "Visual Studio Express 2013 for Windows Desktop" specifically as there are several versions on that page.

# Build wxWidgets

- Download wxWidgets 3.0.2 from https://www.wxwidgets.org/downloads/ and unzip it to somewhere such as C:\Dev\wxWidgets-3.0.2\
- Navigate to C:\Dev\wxWidgets-3.0.2\build\msw\ (or wherever you extracted) and open wx_vc12.sln
- Choose "Release" and "x64" the build configuration and "Build Solution" 
- All should be compiled now, can close project.
     
	 
# Install CMake:
    
Download CMake from:
    http://www.cmake.org/download/
For this guide I've used:
    http://www.cmake.org/files/v3.1/cmake-3.1.1-win32-x86.exe

Just install CMake with default or preferred options.

# Build CubicSDR:

- Clone or download zip from https://github.com/cjcliffe/CubicSDR/ to a folder of your choice.  I'll use C:\Dev\CubicSDR
- Run CMake GUI
- Choose C:\Dev\CubicSDR for source.
- Choose C:\Dev\CubicSDR\cmake_build (or any output folder of your choice)
- Click Configure.
- Choose "Visual Studio 12 2013 Win64" and Finish.
- Note that both DirectSound and WASAPI are supported but WASAPI is enabled by default until some minor issues with DirectSound are resolved (YMMV).
- Set wxWidgets_ROOT_DIR to "C:\Dev\wxWidgets-3.0.2".
- Configure again, all should be good, then Generate.
- Navigate to your output folder (i.e. C:\Dev\CubicSDR\cmake_build\) in explorer and open CubicSDR.sln.
- Once open select "Release" and "x64" build configuration and then run "Build Solution" (F7)
- CubicSDR.exe should now be in the output folder (i.e. C:\Dev\CubicSDR\cmake_build\x64\) and ready to run.

cjcliffe, 2015-01-28:
Build instructions now available in the Wiki -- https://github.com/cjcliffe/CubicSDR/wiki

cjcliffe, 2015-02-06:
@Dantali0n thanks -- I'll update the Wiki

cjcliffe, 2015-03-15:
Thanks for the help here everyone -- the build notes should all be in the Wiki now; please feel free to make any updates or additions there -- I'll close this issue for now.

closed 2015-03-15 by cjcliffe

dovandung, 2015-05-25:
Build on Visual C 2013 with "Release" is Ok , but if you want to build with "Debug" you need some modify : Put Preprocessor _ITERATOR_DEBUG_LEVEL=1 , and on ScopeCanvas::OnPaint(wxPaintEvent& WXUNUSED(event)) you need to modify code , if you access array demodAudioData->data, it will catch up a runtime error .
I change the code as following it work 
 const wxSize ClientSize = GetClientSize();
	std::vector<float> tmp;
	
    if (!wxGetApp().getAudioVisualQueue()->empty()) {
        AudioThreadInput *demodAudioData;
        wxGetApp().getAudioVisualQueue()->pop(demodAudioData);
		int iMax = demodAudioData->data.size();
        if (demodAudioData && iMax) {
            if (waveform_points.size() != iMax * 2) {
                waveform_points.resize(iMax * 2);
            }
		
			tmp.resize(iMax);
			memcpy(&tmp[0], &demodAudioData->data[0], iMax*sizeof(float));
			
			for (int i = 0 ; i < iMax; i++) {
			//for (int i = 0, iMax = demodAudioData->data.size(); i < iMax; i++) {
					
				waveform_points[i * 2 + 1] = tmp[i] * 0.5f;
					waveform_points[i * 2] = ((double)i / (double)iMax);
				
            }
			

            setStereo(demodAudioData->channels == 2);
        } else {
            std::cout << "Incoming Demodulator data empty?" << std::endl;
        }
    }

dovandung, 2015-05-25:
I still checking why it make a runtime error