cjcliffe/CubicSDR #38: Audio Buzz

bodrick, 2015-01-21:
Sometimes when I'm listening I got a noticeable buzz on the audio, it doesn't seem to be consistent, and it will go away if I either restart the program or if I delete the demodulator and add it again.  This is on Windows 8.1 64bit

cjcliffe, 2015-01-22:
I've had this happen a few times myself, I'm investigating using the prototype mixer required for OSX/Linux on Windows as well to reduce the audio thread overhead.

sm0yyq, 2015-01-23:
I do also have problem with the sound, ... on Ubuntu 14.10.
It's sound like buzzing / crackling noise, .. i have used CubicSDR with rtl_rpc and also direct on my computer connected to the dongle.

Same audioproblems. .. 

/M

cjcliffe, 2015-01-23:
@sm0yyq Do you know what chipset your audio device is?  I'm not sure if it's device related but so far it looks like it could be either that or a bug somewhere in the audio code.

sm0yyq, 2015-01-23:
I have a x-fi pci-e

cjcliffe, 2015-01-24:
Hmm, I wonder if doing a build with ASIO and/or DirectSound support would work better for you..  Both work and all can be used together but I've just defaulted to WASAPI -- I'll keep that in mind when preparing the bins as others might like to compare and choose the best for their device.

sm0yyq, 2015-01-24:
Hmm, .. now i use Ubuntu, . so i guess ASIO and DirectSound is not an option .. :)

sm0yyq, 2015-01-24:
Do u meen bodrick now, ??, .. cose i use Ubuntu and not windows. :)

bodrick, 2015-01-24:
So I've been experimenting and I've noticed that I also get random issues with sound.
Issue 1) If I delete an existing modulator, sometimes the app will crash hard - I'm assuming its a threading issue
issue 2) Sometimes when I've created a modulator then deleted it, I can't add another one
Issue 3) Occasionally I will create a modulator but get no sound, to fix I have to restart the app.

Not sure if this helps much, but I figured its a start, I can try and see if I can build for DirectSound / ASIO and see if that makes any difference.

cjcliffe, 2015-01-25:
@bodrick 
1) This should be fixed in the latest master, there were some serious de-init issues being masked by a thread not actually exiting.
2) This is a copy-last-demodulator context issue, usually holding SHIFT will allow a new one to be added; not sure of the source of the issue yet.
3) Using another sound driver might help this as my mileage varies based on output device; I'll try to make a build with ASIO and DS for testing.

@sm0yyq Yes sorry, I've mostly experienced this in Windows but rarely -- On Linux or OSX I'm betting it's somewhere in my audio code where I wrote a rudimentary mixer since I couldn't open multiple instances of the same device with CoreAudio -- Linux had the same problem so I enabled the mixer there.   It also defaults to pulse audio at the moment; but you should be able to choose OSS or ALSA as part of the CMake config with USE_AUDIO_ALSA=1 and USE_AUDIO_OSS=1.  Optionally it should work with Jack via USE_AUDIO_JACK=1.  If you want to try disabling the mixer there's a define for USE_MIXER in the AudioThread.h header.

bodrick, 2015-01-25:
@cjcliffe So I actually just compiled using DS and it seems to be much more reliable and stable, I didn't seem to run into the problem I was running in when compiled using WSAPI.

cjcliffe, 2015-01-25:
@bodrick based on what I've read DS was using WASAPI under the hood either way; but perhaps the vendor DirectSound drivers are better somehow.  Glad to hear that helps.

sm0yyq, 2015-01-26:
@cjcliffe I did try to use ALSA instead, and it didnt remove the bad sound :/. I also tried to remove the Mixer and it didnt made any changes. So i dont realy know whats causing the problem. One thing i have not tried yet is to connect my dongle directly to my comp, .. so i will and get back to u.

cjcliffe, 2015-02-06:
@bodrick I've made DS the default now as I've resolved the issues I had with it, it seems more stable for me and I haven't ran into any issues yet.   Issue 2) from your original request has returned and now persists but I'm working on that now.

cjcliffe, 2015-02-06:
@bodrick that was easy -- Issue 2) now fixed in master

cjcliffe, 2015-02-09:
@sm0yyq check out https://github.com/cjcliffe/CubicSDR/commit/add04883eb532f9458cd16df39b1a9a1ca466172 I made some adjustments to the mixer that should possibly fix clipping on devices that don't natively support floating point output.  I noticed that some devices were clipping by default and sounded like a buzzing/crackling noise or broken speaker.

Let me know if that helps!

cjcliffe, 2015-02-12:
'mixer-fixes' branch has been merged to master along with some other audio oddities that were fixed; hoping that will help with this issue.

cjcliffe, 2015-03-15:
I can no longer reproduce the buzzing audio issues I had in Linux here; please re-open issue if problem persists.

closed 2015-03-15 by cjcliffe