cjcliffe/CubicSDR #52: Vertical Sync -- Excessively high framerates

cjcliffe, 2015-02-12:
Some platforms may not currently perform vertical sync correctly if an available "SwapInterval" extension is not detected and used.  If you're experiencing excessive frame rates please post your system information (CPU/GPU/OS) here.

mryndzionek, 2015-02-15:
System specs:
```sh
CPU: Intel Core i3-4330
GPU: Xeon E3-1200 v3/4th Gen Core Processor Integrated Graphics Controller
OS: Linux Mint LMDE
```
Running at refresh rate 60Hz the CPU load is about 75% without demodulator (85% with demodulator).
Running with `vblank_mode=0` the CPU load is 95% without demodulator (115% with demodulator)
No vertical sync swap function is detected.

cjcliffe, 2015-02-16:
With the waterfall updates it actually runs (sans demodulator of course) on my old Intel Atom (~2007) 1.6Ghz netbook with an intel GMA950 chipset -- which appears to be using the same driver series as yours (their HD graphics series and beyond are successors of the GMA)

I'll do some tinkering; since what's available on the GMA950 may still be available on the newer generations.

mryndzionek, 2015-02-17:
I've installed new Linux Mint (kernel `3.13.0-37`) and `glxinfo` reports the swap extensions. Dlsym does not work but `glXGetProcAddressARB` seems to locate the functions:
```sh
Available vertical sync SwapInterval functions: 
	glxSwapIntervalEXT: No
	glxSwapIntervalMESA: Yes
	glxSwapIntervalSGI: Yes

```
Now however the frame rate is very low (probably [this](https://github.com/cjcliffe/CubicSDR/issues/30) issue).

cjcliffe, 2015-02-18:
interesting, can you disable window manager/driver vsync?  That usually fixes it for me as the glxSwapInterval (to my knowledge) doesn't depend on vsync being enabled it just synchronizes individual swaps to it; but window manager vsync will wait for each window to draw in series.

I believe there's a way to synchronize swaps between contexts so it's not an issue but I haven't gotten into that yet.

mryndzionek, 2015-02-21:
Running with `vblank_mode=1` seems to fix this problem. I don't see the option to disable windows manager/intel driver vsync anywhere.

cjcliffe, 2015-02-24:
Interesting.. I wonder if I can just set that environment variable myself at the start of CubicSDR.cpp app initialization and it will carry into the OpenGL context.....

closed 2015-05-31 by cjcliffe