cjcliffe/CubicSDR #54: OpenSUSE issues

cjcliffe, 2015-02-12:
Reported via e-mail: 

I've just finished some initial packaging for openSUSE:
https://build.opensuse.org/package/show/home:mnhauke:sdr/CubicSDR

I've found some minor issues...
* cmake does not check for some requirements
 - liquid-sdr
 - rtl-sdr
 - wx-widgets >= 3.0 (it first found my old 2.0 headers wich ended up
in some compile errors
 - libpulse-devel
 - probably other libraries too

* there's actually no "make install" in the buildsystem
 - CubicSDR binary and fonts get copied to dir $arch
 - the fonts have are only read from the current dir
 - see my ugly workarounds in the specfile.

croutonage, 2015-04-13:
>>wx-widgets >= 3.0 (it first found my old 2.0 headers wich ended up in some compile errors

Had this issue too.. it needs to be 3.0 , but cmake will accept 2.8, then give compile errors. Perhaps easily fixed by editing the readme file/build instructions

camconn, 2015-08-22:
On OpenSUSE 13.2 I had to install these packages:
- `bitstream-vera-fonts`
- `automake`
- `autoconf`
- `cmake`
- `wxwidgets-3_0`
- `wxwidgets-3_0-devel
- `rtaudio-devel`
- `rtaudio-debugsource`
- `librtaudio4`
- `librtlsdr0`
- `libosmosdr0`

I had to compile the following packages from scratch, as there were OpenSUSE packages:
- Liquid DSP (https://github.com/jgaeddert/liquid-dsp)

I didn't install these things at all:
- LodePNG (http://lodev.org/lodepng/)
- BMFont (http://www.angelcode.com/products/bmfont/)

cjcliffe, 2016-02-26:
Can't find a reliable solution to the final wxWidgets issue here; going to handle it with documentation.

closed 2016-02-26 by cjcliffe