cjcliffe/CubicSDR #56: FM stereo

jacobseptember, 2015-02-21:
Hello! Thanks for the program; it took a bit of effort to install it, but wasn't too bad.

About stereo FM reception: This is listed as "checked" in the README file. However, by testing with my iPod and a small FM transmitter, I can tell that this "stereo" mode (enabled with the spacebar) seems to just add a delay on one channel: Even with the iPod panned all the way left or right there is sound coming out of both speakers, only with a weird echo effect. (This is rather disorienting at times!)

I'm not sure if this is an issue with CubicSDR or with my dongle—I'm using a generic RTL2832U. This is the first SDR-receiver software I've used that I'm *certain* is supposed to receive in stereo, so I'm not sure my hardware can do stereo. I don't see why it shouldn't, though.

cjcliffe, 2015-02-24:
Stereo isn't really a feature of the dongle; it's part of the audio much like Dolby or any other encoding -- the stereo channel is just pushed out of the range of human hearing or receiver range; it's up to the software or receiver to mix that signal down to make it audible.   It sounds like your transmitter might actually be Mono; do you see the 3 trails that typically indicate stereo on the waterfall (one on either side of the centre) when the output is quiet?

That said though; the FM Stereo is experimental -- it currently doesn't even check if there's actually a stereo pilot tone it just does the steps to mix-down the L-R channel and use it to split the L+R channel and apply some basic filtering (so you can use it on signals without stereo to cause interesting effects)

If you use it on a FM Mono station there will be just be a minor delay effect caused by the shift; if it's narrow FM or SSB/AM you'll get a weird distortion or delay echo since the narrow bandwidth causes some interesting feedback in the stereo code.

As it is now I also haven't added a proper phase-lock to the pilot tone for mixing-down the L-R so there's likely some random phasing/panning artefacts going on -- but it is a functional FM Stereo implementation.

f1vefour, 2015-02-24:
The RTL2832 does support stereo.

jacobseptember, 2015-02-25:
The transmitter is in fact transmitting stereo. Here's a demonstration of what I'm seeing (detailed commentary in the description): https://www.youtube.com/watch?v=9WgXuHogOmg

Perhaps it is a problem with wide/narrowband? But it doesn't look like I'm missing much of the output from what I can see of the waterfall.

jacobseptember, 2015-02-25:
Also, I just want to say: I don't know how your SDR software is different from other peoples', but it it FAR, FAR better about not having interference depending on where my body is relative to the antenna/whether I'm touching the dongle/etc. I don't know why this would be different for the same physical setup, but it is—and CubicSDR is far less sensitive than anything else I've used.

f1vefour, 2015-02-25:
From the video it appears that your issue is from latency caused by high CPU usage.

jacobseptember, 2015-02-25:
> latency caused by high CPU usage

I don't think so. The excessive CPU usage shown in the video was due to the act of recording the screen;  CubicSDR uses ~50% CPU out of 400% (four-core i5) with the waterfall, and ~10% out of 400% with the window resized so the waterfalls cannot be seen (another aspect that I very much appreciate). The wobble occurs in both cases.

One other interesting factor: If the iPhone is panned center and I enable the "mono audio" setting, the wobble disappears again. Keeping mono enabled and panning left or right brings it back.

cjcliffe, 2015-02-26:
@jacobseptember -- Thanks for the video; it appears to be the result of the lack of phase lock to the pilot in my code causing lots of panning artifacts; I'll make improvements to it and can even use a similar iPod attachment I have here for testing -- It was mostly experimental so far and definitely needs work!

cjcliffe, 2015-02-26:
@jacobseptember In regards to the interference it's likely that it's being filtered out by the automatic spectrum view adjustments, if the changes are gradual you won't notice overall noise floor changes much as it's constantly adjusting to match.   I'm betting once I add the faint background lines to represent dB levels you'll see them dip and rise gradually as you move.

jacobseptember, 2015-02-26:
Re: interference:
I don't know about specific dB levels; I'm just talking about audible white noise/static. I'd been using the `rtl_fm` command-line tool, and it would be very finicky. I'd be standing in my room and shift my body slightly, and the signal would be overcome by static. I can stand in the same place using CubicSDR and shift all around with absolutely no ill effects. It probably is Liquid DSP—I didn't have to install that for `rtl_fm`.

cjcliffe, 2015-06-06:
@jacobseptember Improved FM support is now merged to master, let me know what you think when you get a chance to try it out; will be included in 0.1.3 build soon as well.  Note that there will always be a tiny bit of bleed from L to R due to the nature of stereo FM modulation but it should be greatly improved now.

closed 2015-06-06 by cjcliffe

jacobseptember, 2015-07-26:
Sorry I haven't been able to get back to this in a while—I was traveling, and then I was busy. But it sounds good! I thought I could detect some strange volume ups-and-downs when I tested "California Dreamin'" again—the little cymbal-crashes seemed to coincide with a dampening effect on the other sounds. But I could be making that up.

Listening to the actual radio sounds great.