cjcliffe/CubicSDR #63: Enhancement Request: More frequency input methods

madboom, 2015-03-17:
I like the variable speed tuning scrubber. It would also be great if the current center frequency was displayed larger where each digit can be incremented/decremented by clicking the upper/lower half of each digit or mouse hovering over a digit allows inc/dec by scrolling the mouse wheel. It would also be great to have a standby frequency "scratchpad" where the freq. can be entered from the keyboard, and then set by either hitting enter, or clicking a commit or set button. If any of these features are already planned or implemented, please disregard.

cjcliffe, 2015-03-18:
Thanks!  I definitely need to implement something like the familiar click-to-tune frequency as it's widely accepted and understood.  

I have a plan in my head for a universal bookmark / history / search popup which will be similar to a scratch pad idea -- I'm thinking space-bar as the trigger and enter to accept plus an area below the search box with some clickable results from bookmarks and history.

In addition I'm thinking the search box will accept suffixed and delimited frequencies e.g. "1.09Ghz", "100Mhz", "14,000khz", etc..

cjcliffe, 2015-04-03:
the new hybrid scrubbing/frequency dial type tuning bar is now committed to master; 0.1.2 will be released with it included soon.

mmgarland3, 2015-04-03:
can we get mousewheel support too?  new build works great but I keep intuitively trying to scroll the digits up/down.

cjcliffe, 2015-04-03:
mousewheel support should be possible, right now the mousewheel is globally controlling the waterfall/spectrum zoom -- I'll see what I can do to separate it though.

cjcliffe, 2015-04-08:
@mmgarland3 been super busy this past week but I had a moment to try implementing mouse wheel support and it's now committed to master (was a fairly simple change); check it out and let me know if it's what you're looking for.

mmgarland3, 2015-04-08:
Yes, that works great!  Perfect, thanks!

cjcliffe, 2015-05-04:
Status:
[x] Variable speed scrubbing
[x] Large incremental digits
[x] Mouse wheel support
[ ] Direct frequency input via keyboard

cjcliffe, 2015-05-11:
direct frequency input via keyboard is now merged to master, see pull #78 for basic instructions

closed 2015-05-11 by cjcliffe