cjcliffe/CubicSDR #66: 3.2M Input rate with low bandwidth demod exceeds liquid-dsp 10 stage resampler limit

cjcliffe, 2015-03-26:
Need to add some half-band re-samplers that can reduce the input load before the arbitrary re-sampler stage to prevent it from exceeding 10 stages.

croutonage, 2015-04-13:
Don't think this samplerate is attainable for most people. 2.88M is the max for me without errors. Never tried on apple though.

rtl_test -s 3200000
Found 1 device(s):
  0:  Generic, RTL2832U DVB-T, SN: 4000xxFF115

Using device 0: Generic RTL2832U
Found Elonics E4000 tuner
Supported gain values (14): -1.0 1.5 4.0 6.5 9.0 11.5 14.0 16.5 19.0 21.5 24.0 29.0 34.0 42.0 
Sampling at 3200000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
lost at least 2880 bytes
lost at least 2848 bytes
lost at least 2280 bytes
lost at least 3612 bytes

cjcliffe, 2015-04-14:
@croutonage I have been able to get it to work on a pair of Newsky-type 820T2 dongles here at 3.2M while using a USB3 port on my desktop and no hub -- this issue will be far more relevant when support is added for higher bandwidth devices though.

cjcliffe, 2015-10-21:
liquid-dsp multi-stage resampler now supports 16 half-band stages which is more than the latest soapy builds should ever need.

closed 2015-10-21 by cjcliffe