cjcliffe/CubicSDR #68: PPM correction

cjcliffe, 2015-04-03:
Ability to configure device PPM
 - Simple menu option like offset
 - Basic configuration dialog for devices that can be saved and auto-loaded
 - Some sort of PPM correction helper?

mmgarland3, 2015-04-03:
Don't see any reason to put on tuning bar; maybe on the right end of the status bar, if at all?

For a helper, would need a known freq to lock carrier and adjust from there - I use the local airport ATIS freq to calibrate.

I think there's a project called kalibrate_sdr or something like that that does this... Let me research and get back to you.

croutonage, 2015-04-13:
Passing as a commandline arg would be useful as a first step

Setting it from inside the program would be nice, crucial to be able to do it while viewing the waterfall and zoomed in, if so, automatically deriving it from the (programmable) device's serial number would be pretty cool. take note some ppm values are negative however. 

For now, I have simply forked and hardcoded it in as an extra variable. It is a register setting within the device, so there is no need to calculate anything with regard to displayed freqs etc.

cjcliffe, 2015-04-14:
I've started progress on having it save and load the PPM correction values per-device from a generic xml configuration file configured in the default user data app dir.

First pass I'll just had a simple dialog input like the current frequency offset and it'll save to config and load automatically on start-up.   Second iteration will likely have some sort of UI (possibly a toggle button on the tuning bar) that lets you adjust the PPM using the new tuning controls; and later the utility mode could have some additional tools to aid in automatic calibration.

cjcliffe, 2015-04-24:
PPM can now be adjusted via the menu option or by hovering the frequency tuning bar while holding ALT (or Meta+Alt for me in Debian).    PPM per-device will be saved and loaded each time and applied automatically when switching between multiple devices.

closed 2015-04-24 by cjcliffe