cjcliffe/CubicSDR #72: GUI does not spawn when run from outside its directory

SpiritQuaddicted, 2015-04-25:
I build CubicSDR and ran it like "x64/CubicSDR", then I wondered why no GUI showed up. The GUI only shows if I cd into x64/ and run the binary there. It would be nice to issue a warning or error.

cjcliffe, 2015-04-25:
The CPack generator for Linux, Windows and OSX can produce installation packages that don't have this issue -- but it would be good to add a check and possibly change the working directory in this case -- I find I have to configure my IDE run path each time I regenerate the project due to the same issue.

cjcliffe, 2015-06-04:
Font loader will now try the executable path if the font isn't found.  Makes it decent now for building with eclipse and others without configuration since they don't default to the x64 subdir.

Thanks!

closed 2015-06-04 by cjcliffe