cjcliffe/CubicSDR #74: Crash when closing on OS X 10.10.3

Toontje, 2015-04-27:
Process:               CubicSDR [51649]
Path:                  /Applications/CubicSDR.app/Contents/MacOS/CubicSDR
Identifier:            com.cubicproductions.cubicsdr
Version:               ??? (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           CubicSDR [51649]
User ID:               501

Date/Time:             2015-04-27 12:22:03.896 +0200
OS Version:            Mac OS X 10.10.3 (14D136)
Report Version:        11
Anonymous UUID:        F677BE74-5929-469A-253F-12C0D18FBFD7

Sleep/Wake UUID:       317D490E-6760-4FCE-A38E-0ABC7E07FE47

Time Awake Since Boot: 430000 seconds
Time Since Wake:       6000 seconds

Crashed Thread:        4

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Application Specific Information:
terminating with uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
abort() called

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_pthread.dylib       	0x00007fff935b05a5 pthread_mutex_lock + 2
1   libsystem_c.dylib             	0x00007fff93ce38d9 __cxa_finalize_ranges + 334
2   libsystem_c.dylib             	0x00007fff93ce3bd0 exit + 55
3   libdyld.dylib                 	0x00007fff925535d0 start + 8

Thread 1:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff8dd92232 kevent64 + 10
1   libdispatch.dylib             	0x00007fff92911a6a _dispatch_mgr_thread + 52

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff8dd8c4de mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff8dd8b64f mach_msg + 55
2   com.apple.CoreFoundation      	0x00007fff94485eb4 __CFRunLoopServiceMachPort + 212
3   com.apple.CoreFoundation      	0x00007fff9448537b __CFRunLoopRun + 1371
4   com.apple.CoreFoundation      	0x00007fff94484bd8 CFRunLoopRunSpecific + 296
5   com.apple.AppKit              	0x00007fff87e7866b _NSEventThread + 137
6   libsystem_pthread.dylib       	0x00007fff935b2268 _pthread_body + 131
7   libsystem_pthread.dylib       	0x00007fff935b21e5 _pthread_start + 176
8   libsystem_pthread.dylib       	0x00007fff935b041d thread_start + 13

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff8dd91136 __psynch_cvwait + 10
1   com.cubicproductions.cubicsdr 	0x00000001016bdee8 RtApiCore::stopStream() + 136
2   com.cubicproductions.cubicsdr 	0x000000010168a6f9 AudioThread::threadMain() + 985
3   com.cubicproductions.cubicsdr 	0x0000000101686251 void* std::__1::__thread_proxy<std::__1::tuple<void (AudioThread::*)(), AudioThread*> >(void*) + 113
4   libsystem_pthread.dylib       	0x00007fff935b2268 _pthread_body + 131
5   libsystem_pthread.dylib       	0x00007fff935b21e5 _pthread_start + 176
6   libsystem_pthread.dylib       	0x00007fff935b041d thread_start + 13

Thread 4 Crashed:
0   libsystem_kernel.dylib        	0x00007fff8dd91286 __pthread_kill + 10
1   libsystem_c.dylib             	0x00007fff93ce2b53 abort + 129
2   libc++abi.dylib               	0x00007fff92960a21 abort_message + 257
3   libc++abi.dylib               	0x00007fff929889b9 default_terminate_handler() + 243
4   libobjc.A.dylib               	0x00007fff8cb1c7eb _objc_terminate() + 124
5   libc++abi.dylib               	0x00007fff929860a1 std::__terminate(void (*)()) + 8
6   libc++abi.dylib               	0x00007fff92985b30 __cxa_throw + 121
7   libc++.1.dylib                	0x00007fff8f5e6b25 std::__1::__throw_system_error(int, char const*) + 77
8   com.cubicproductions.cubicsdr 	0x0000000101682777 DemodulatorThread::threadMain() + 7543
9   libsystem_pthread.dylib       	0x00007fff935b2268 _pthread_body + 131
10  libsystem_pthread.dylib       	0x00007fff935b21e5 _pthread_start + 176
[... 398 lines cut ...]
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1333 MHz, 0x0000, -
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1333 MHz, 0x0000, -
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xD6), Broadcom BCM43xx 1.0 (5.106.98.100.24)
Bluetooth: Version 4.3.4f4 15601, 3 services, 18 devices, 1 incoming serial ports
Network Service: AirPort, AirPort, en1
Serial ATA Device: OWC Mercury Electra 6G SSD, 480,1 GB
Serial ATA Device: MATSHITADVD-R   UJ-8A8
USB Device: FaceTime HD Camera (Built-in)
USB Device: Hub
USB Device: BRCM2070 Hub
USB Device: Bluetooth USB Host Controller
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Hub
USB Device: USB2.0 Hub
USB Device: Usb production
USB Device: USB2.0 Hub
USB Device: Samsung S2 Portable
USB Device: Hub
USB Device: CP2102 USB to UART Bridge Controller
USB Device: USB Audio CODEC
USB Device: USB2.0 Hub
USB Device: RTL2838UHIDIR
USB Device: Jabra UC VOICE 750a MS
USB Device: IR Receiver
Thunderbolt Bus: MacBook Pro, Apple Inc., 22.1

cjcliffe, 2015-05-31:
latest commit to master may correct this issue which I believe is related to #77

cjcliffe, 2015-06-03:
0 crashes since update on my macs; going to close this until further notice :+1:

closed 2015-06-03 by cjcliffe