cjcliffe/CubicSDR #77: Crash *** Error in `./CubicSDR': double free or corruption (fasttop): 0x0000000002e95af0 ***

UbhaSecurity, 2015-05-05:
Demodulator preprocessor thread started..
Set sample rate: 1000000
Exact sample rate is: 1000000.026491 Hz
Terminating demodulator audio thread..
Terminating demodulator thread..
Terminating demodulator preprocessor thread..
Audio thread done.
Demodulator preprocessor thread done.
Demodulator thread done.
Demodulator worker thread done.
Garbage collected demodulator instance 261.500
Set sample rate: 2880000
Set sample rate: 2500000
Exact sample rate is: 2500000.107620 Hz
Set sample rate: 1000000
Exact sample rate is: 1000000.026491 Hz
Set sample rate: 3200000
Set sample rate: 2000000
Exact sample rate is: 2000000.052982 Hz
Set sample rate: 2500000
Exact sample rate is: 2500000.107620 Hz
Terminating SDR thread..
SDR thread done.
Terminating SDR post-processing thread..
SDR post-processing thread done.
Terminating demodulator audio thread..
Terminating demodulator thread..
Terminating demodulator preprocessor thread..
Terminating demodulator audio thread..
Demodulator worker thread done.
*** Error in `./CubicSDR': double free or corruption (fasttop): 0x0000000002e95af0 ***
Audio thread done.
Audio thread done.
Terminating demodulator thread..
Terminating demodulator preprocessor thread..
Aborted

Linux xbmc 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt9-3~deb8u1 (2015-04-24) x86_64 GNU/Linux

UbhaSecurity, 2015-05-06:
Not reproduce-able aik.
So no showstopper or usual crash.

Use Cubicsdr for hours daily with no issues.

cjcliffe, 2015-05-10:
yeah I think this is just an old one I re-introduced along with #74 -- should be a quick fix

cjcliffe, 2015-05-31:
latest commit to master may correct this issue; it appears to have fixed it on OSX for me

cjcliffe, 2015-06-03:
0 crashes since update on my linux box; going to close this until further notice :+1:

closed 2015-06-03 by cjcliffe