cjcliffe/CubicSDR #81: Some bug of program

dovandung, 2015-05-26:
When compile program for Windows with "Debug" option , program crash after start decode audio and it says runtime error.Finding more, I see if you remove line code in ScopeCanvas.cpp "waveform_points[i * 2 + 1] = demodAudioData->data[i] * 0.5f;" , it will not crash anymore. So I think there is something wrong in this code . The second finding : If you compile with "Release" It runs but I see on audio waveform there are a lot spike noise .
Finally , I detect that there is a conflict in access data memory of AudioThreadInput class . During you read out data of audioVisOutput , the DemodulatorThread is writing data in , so it flag an error if you run in debug mode and AudioScope has many spike .
I correct program by using a busy_readwrite_flag on threads DemodulatorThread and ScopeCanvas as following 
On DemodulatorThread : 
   while (busy_readwrite_flag ) ;
       busy_readwrite_flag = 1 ;
   // write data
      busy_readwrite_flag = 0 ;
Same On ScopeCanvas
   while (busy_readwrite_flag ) ;
       busy_readwrite_flag = 1 ;
   // get data
      busy_readwrite_flag = 0 ;
After correct this program run fine on Debug mode and AudioWave form there is no spike glitch .
Because this program there are many Thread run at same times , I think still have some more bugs like this . Because after correct this program only run with _ITERATOR_DEBUG_LEVEL = 1 , if you set 
_ITERATOR_DEBUG_LEVEL = 2 program still has run-time error

cjcliffe, 2015-05-26:
Good find, thank you! 

I may be passing the same memory to both or forgetting to increment a reference counter.  It may be incorrectly copying a pointer somewhere instead of the data -- the visual and audio should be completely separate memory.   This type of code is used in a few areas that I will have to investigate -- if your read/write flag method works I may adopt that.

I will put priority on fixing these -- thanks for the debugging tips!

dovandung, 2015-05-27:
I research more and on Windows version I modify as following and It works perfect with "DEBUG" compilation , no runtime error. To correct memory conflict when ScopeCanvas read data 
1/ Add variable in class AudioThreadInput 
class AudioThreadInput: public ReferenceCounter {
public:
    long long frequency;
    int sampleRate;
    int channels;
    float peak;
    std::vector<float> data;
	std::mutex busy_update;
    AudioThreadInput() :
            frequency(0), sampleRate(0), channels(0), peak(0) {

    }

    ~AudioThreadInput() {
        std::lock_guard < std::mutex > lock(m_mutex);
    }
};
2/ Modify code in ScopeCanvas and SDRPostThread 
a/In ScopeCanvas :

if (iqData && iqData->data.size()) {
			iqData->busy_rw.lock();
            setData(iqData);
            if (otherWaterfallCanvas) {
                otherWaterfallCanvas->setData(iqData);
            }
			iqData->busy_rw.unlock();
        } else {
            std::cout << "Incoming IQ data empty?" << std::endl;
        }
    }

b/ In SDRPostThread :

 if (iqVisualQueue.load() != NULL && iqVisualQueue.load()->empty()) {
				visualDataOut->busy_rw.lock();
                if (visualDataOut->data.size() < num_vis_samples) {
                    if (visualDataOut->data.capacity() < num_vis_samples) {
                        visualDataOut->data.reserve(num_vis_samples);
                    }
                    visualDataOut->data.resize(num_vis_samples);
                }

                visualDataOut->frequency = data_in->frequency;
                visualDataOut->sampleRate = data_in->sampleRate;
                visualDataOut->data.assign(dataOut.begin(), dataOut.begin() + num_vis_samples);
				visualDataOut->busy_rw.unlock();
                iqVisualQueue.load()->push(visualDataOut);
            }
There is another bug same type with this bug in WaterfallCanvas  and this is modify :
1/Add a variable 

class DemodulatorThreadIQData: public ReferenceCounter {
public:
    long long frequency;
    long long sampleRate;
	std::mutex busy_rw;
    std::vector<liquid_float_complex> data;

    DemodulatorThreadIQData() :
            frequency(0), sampleRate(0) {

    }

    ~DemodulatorThreadIQData() {

    }
};

2/ Modify code in SDRPostThread 

    if (iqVisualQueue.load() != NULL && iqVisualQueue.load()->empty()) {
            				visualDataOut->busy_rw.lock();
                if (visualDataOut->data.size() < num_vis_samples) {
                    if (visualDataOut->data.capacity() < num_vis_samples) {
                        visualDataOut->data.reserve(num_vis_samples);
                    }
                    visualDataOut->data.resize(num_vis_samples);
                }

                visualDataOut->frequency = data_in->frequency;
                visualDataOut->sampleRate = data_in->sampleRate;
                visualDataOut->data.assign(dataOut.begin(), dataOut.begin() + num_vis_samples);
				                visualDataOut->busy_rw.unlock();
                iqVisualQueue.load()->push(visualDataOut);
3/ Modify code in WaterfallCanvas
 if (iqData && iqData->data.size()) {
            			iqData->busy_rw.lock();
            setData(iqData);
            if (otherWaterfallCanvas) {
                otherWaterfallCanvas->setData(iqData);
            }
            			iqData->busy_rw.unlock();
        } else {
            std::cout << "Incoming IQ data empty?" << std::endl;
        }

Now program run OK without runtime error anymore .
But program still have some bug when exit . I will investigate later

dovandung, 2015-05-27:
I research more and on Windows version I modify as following and It works perfect with "DEBUG" compilation , no runtime error. To correct memory conflict when ScopeCanvas read data 1/ Add variable in class AudioThreadInput class AudioThreadInput: public ReferenceCounter {public:    long long frequency;    int sampleRate;    int channels;    float peak;    std::vector<float> data; std::mutex busy_update;    AudioThreadInput() :            frequency(0), sampleRate(0), channels(0), peak(0) {
    }
    ~AudioThreadInput() {        std::lock_guard < std::mutex > lock(m_mutex);    }};2/ Modify code in ScopeCanvas and SDRPostThread a/In ScopeCanvas :
if (iqData && iqData->data.size()) { iqData->busy_rw.lock();            setData(iqData);            if (otherWaterfallCanvas) {                otherWaterfallCanvas->setData(iqData);            } iqData->busy_rw.unlock();        } else {            std::cout << "Incoming IQ data empty?" << std::endl;        }    }
b/ In SDRPostThread :
 if (iqVisualQueue.load() != NULL && iqVisualQueue.load()->empty()) { visualDataOut->busy_rw.lock();                if (visualDataOut->data.size() < num_vis_samples) {                    if (visualDataOut->data.capacity() < num_vis_samples) {                        visualDataOut->data.reserve(num_vis_samples);                    }                    visualDataOut->data.resize(num_vis_samples);                }
                visualDataOut->frequency = data_in->frequency;                visualDataOut->sampleRate = data_in->sampleRate;                visualDataOut->data.assign(dataOut.begin(), dataOut.begin() + num_vis_samples); visualDataOut->busy_rw.unlock();                iqVisualQueue.load()->push(visualDataOut);            }There is another bug same type with this bug in WaterfallCanvas  and this is modify :
1/Add a variable 
 class DemodulatorThreadIQData: public ReferenceCounter {public:    long long frequency;    long long sampleRate; std::mutex busy_rw;    std::vector<liquid_float_complex> data;
    DemodulatorThreadIQData() :            frequency(0), sampleRate(0) {
    }
    ~DemodulatorThreadIQData() {
    }};
2/ Modify code in SDRPostThread 
    if (iqVisualQueue.load() != NULL && iqVisualQueue.load()->empty()) {             visualDataOut->busy_rw.lock();                if (visualDataOut->data.size() < num_vis_samples) {                    if (visualDataOut->data.capacity() < num_vis_samples) {                        visualDataOut->data.reserve(num_vis_samples);                    }                    visualDataOut->data.resize(num_vis_samples);                }
                visualDataOut->frequency = data_in->frequency;                visualDataOut->sampleRate = data_in->sampleRate;                visualDataOut->data.assign(dataOut.begin(), dataOut.begin() + num_vis_samples);                 visualDataOut->busy_rw.unlock();                iqVisualQueue.load()->push(visualDataOut);3/ Modify code in WaterfallCanvas if (iqData && iqData->data.size()) {             iqData->busy_rw.lock();            setData(iqData);            if (otherWaterfallCanvas) {                otherWaterfallCanvas->setData(iqData);            }             iqData->busy_rw.unlock();        } else {            std::cout << "Incoming IQ data empty?" << std::endl;        }
Now program run OK without runtime error any more .But program still have some bug when exit . I will investigate later

dovandung, 2015-05-27:
Sorry previous attach wrong code , I correct in this email .I research more and on Windows version I modify as following and It works perfect with "DEBUG" compilation , no runtime error. To correct memory conflict when ScopeCanvas read data 1/ Add variable in class AudioThreadInput class AudioThreadInput: public ReferenceCounter {public:    long long frequency;    int sampleRate;    int channels;    float peak;    std::vector<float> data; std::mutex busy_update;    AudioThreadInput() :            frequency(0), sampleRate(0), channels(0), peak(0) {
    }
    ~AudioThreadInput() {        std::lock_guard < std::mutex > lock(m_mutex);    }};2/ Modify code in ScopeCanvas and SDRPostThread a/In ScopeCanvas :   if (demodAudioData && iMax) {            if (waveform_points.size() != iMax * 2) {                waveform_points.resize(iMax * 2);            }  demodAudioData->busy_update.lock() ; //Dung modify   for (int i = 0 ; i < iMax; i++) { //for (int i = 0, iMax = demodAudioData->data.size(); i < iMax; i++) {   waveform_points[i * 2 + 1] = demodAudioData->data[i] * 0.5f; waveform_points[i * 2] = ((double)i / (double)iMax);             } demodAudioData->busy_update.unlock();             setStereo(demodAudioData->channels == 2);        } else {            std::cout << "Incoming Demodulator data empty?" << std::endl;        }
b/ In DemodulatorThread :

if (ati && audioVisOutputQueue != NULL && audioVisOutputQueue->empty()) { // while (ati_vis->busy_update); ati_vis->busy_update.lock(); //Dung modify             int num_vis = DEMOD_VIS_SIZE;            if (stereo) {                ati_vis->channels = 2;                int stereoSize = ati->data.size();                if (stereoSize > DEMOD_VIS_SIZE) {                    stereoSize = DEMOD_VIS_SIZE;                }
                ati_vis->data.resize(stereoSize);                 for (int i = 0; i < stereoSize / 2; i++) {                    ati_vis->data[i] = ati->data[i * 2];                    ati_vis->data[i + stereoSize / 2] = ati->data[i * 2 + 1];                }            } else {                ati_vis->channels = 1;                if (numAudioWritten > bufSize) {
                    if (num_vis > numAudioWritten) {                        num_vis = numAudioWritten;                    }                    ati_vis->data.assign(resampledOutputData.begin(), resampledOutputData.begin() + num_vis);                } else {                    if (num_vis > bufSize) {                        num_vis = bufSize;                    }                    ati_vis->data.assign(demodOutputData.begin(), demodOutputData.begin() + num_vis);                }
//            std::cout << "Signal: " << agc_crcf_get_signal_level(agc) << " -- " << agc_crcf_get_rssi(agc) << "dB " << std::endl;            } //ati_vis->busy_update = 0; ati_vis->busy_update.unlock();            audioVisOutputQueue->push(ati_vis);        }There is another bug same type with this bug in WaterfallCanvas  and this is modify :
1/Add a variable 
class DemodulatorThreadIQData: public ReferenceCounter {public:    long long frequency;    long long sampleRate; std::mutex busy_rw;    std::vector<liquid_float_complex> data;
    DemodulatorThreadIQData() :            frequency(0), sampleRate(0) {
    }
    ~DemodulatorThreadIQData() {
    }};
2/ Modify code in SDRPostThread 
    if (iqVisualQueue.load() != NULL && iqVisualQueue.load()->empty()) {             visualDataOut->busy_rw.lock();                if (visualDataOut->data.size() < num_vis_samples) {                    if (visualDataOut->data.capacity() < num_vis_samples) {                        visualDataOut->data.reserve(num_vis_samples);                    }                    visualDataOut->data.resize(num_vis_samples);                }
                visualDataOut->frequency = data_in->frequency;                visualDataOut->sampleRate = data_in->sampleRate;                visualDataOut->data.assign(dataOut.begin(), dataOut.begin() + num_vis_samples);                 visualDataOut->busy_rw.unlock();                iqVisualQueue.load()->push(visualDataOut);3/ Modify code in WaterfallCanvas if (iqData && iqData->data.size()) {             iqData->busy_rw.lock();            setData(iqData);            if (otherWaterfallCanvas) {                otherWaterfallCanvas->setData(iqData);            }             iqData->busy_rw.unlock();        } else {            std::cout << "Incoming IQ data empty?" << std::endl;        }
Now program run OK without runtime error any more .But program still have some bug when exit . I will investigate later

dovandung, 2015-05-27:
I research more and on Windows version I modify as following and It works perfect with "DEBUG" compilation , no runtime error. To correct memory conflict when ScopeCanvas read data 
1/ Add variable in class AudioThreadInput 
class AudioThreadInput: public ReferenceCounter {
public:
    long long frequency;
    int sampleRate;
    int channels;
    float peak;
    std::vector<float> data;
    std::mutex busy_update;
    AudioThreadInput() :
            frequency(0), sampleRate(0), channels(0), peak(0) {
    }

    ~AudioThreadInput() {
        std::lock_guard < std::mutex > lock(m_mutex);
    }
};
2/ Modify code in ScopeCanvas and SDRPostThread 
a/In ScopeCanvas :
   if (demodAudioData && iMax) {
            if (waveform_points.size() != iMax * 2) {
                waveform_points.resize(iMax * 2);
            }
			
			demodAudioData->busy_update.lock() ; //Dung modify 
		
			for (int i = 0 ; i < iMax; i++) {
			//for (int i = 0, iMax = demodAudioData->data.size(); i < iMax; i++) {
					
		
				waveform_points[i * 2 + 1] = demodAudioData->data[i] * 0.5f;
					waveform_points[i * 2] = ((double)i / (double)iMax);
				
            }
			demodAudioData->busy_update.unlock();
			
            setStereo(demodAudioData->channels == 2);
        } else {
            std::cout << "Incoming Demodulator data empty?" << std::endl;
        }

b/ In DemodulatorThread :

if (ati && audioVisOutputQueue != NULL && audioVisOutputQueue->empty()) {
			// while (ati_vis->busy_update);
			ati_vis->busy_update.lock(); //Dung modify 
            int num_vis = DEMOD_VIS_SIZE;
            if (stereo) {
                ati_vis->channels = 2;
                int stereoSize = ati->data.size();
                if (stereoSize > DEMOD_VIS_SIZE) {
                    stereoSize = DEMOD_VIS_SIZE;
                }

                ati_vis->data.resize(stereoSize);
				
                for (int i = 0; i < stereoSize / 2; i++) {
                    ati_vis->data[i] = ati->data[i * 2];
                    ati_vis->data[i + stereoSize / 2] = ati->data[i * 2 + 1];
                }
            } else {
                ati_vis->channels = 1;
                if (numAudioWritten > bufSize) {

                    if (num_vis > numAudioWritten) {
                        num_vis = numAudioWritten;
                    }
                    ati_vis->data.assign(resampledOutputData.begin(), resampledOutputData.begin() + num_vis);
                } else {
                    if (num_vis > bufSize) {
                        num_vis = bufSize;
                    }
                    ati_vis->data.assign(demodOutputData.begin(), demodOutputData.begin() + num_vis);
                }

//            std::cout << "Signal: " << agc_crcf_get_signal_level(agc) << " -- " << agc_crcf_get_rssi(agc) << "dB " << std::endl;
            }
			//ati_vis->busy_update = 0;
			ati_vis->busy_update.unlock();
            audioVisOutputQueue->push(ati_vis);
        }
There is another bug same type with this bug in WaterfallCanvas  and this is modify :
1/Add a variable 

class DemodulatorThreadIQData: public ReferenceCounter {
public:
    long long frequency;
    long long sampleRate;
	std::mutex busy_rw;
    std::vector<liquid_float_complex> data;

    DemodulatorThreadIQData() :
            frequency(0), sampleRate(0) {

    }

    ~DemodulatorThreadIQData() {

    }
};

2/ Modify code in SDRPostThread 

    if (iqVisualQueue.load() != NULL && iqVisualQueue.load()->empty()) {
            				visualDataOut->busy_rw.lock();
                if (visualDataOut->data.size() < num_vis_samples) {
                    if (visualDataOut->data.capacity() < num_vis_samples) {
                        visualDataOut->data.reserve(num_vis_samples);
                    }
                    visualDataOut->data.resize(num_vis_samples);
                }

                visualDataOut->frequency = data_in->frequency;
                visualDataOut->sampleRate = data_in->sampleRate;
                visualDataOut->data.assign(dataOut.begin(), dataOut.begin() + num_vis_samples);
				                visualDataOut->busy_rw.unlock();
                iqVisualQueue.load()->push(visualDataOut);
3/ Modify code in WaterfallCanvas
 if (iqData && iqData->data.size()) {
            			iqData->busy_rw.lock();
            setData(iqData);
            if (otherWaterfallCanvas) {
                otherWaterfallCanvas->setData(iqData);
            }
            			iqData->busy_rw.unlock();
        } else {
            std::cout << "Incoming IQ data empty?" << std::endl;
        }

Now program run OK without runtime error any more .
But program still have some bug when exit . I will investigate later

cjcliffe, 2015-05-28:
thanks for the detailed patch report; Ill give this a go -- the exit issue is a double-free of a pointer in the audio thread I believe -- I fixed it once before but think It was reintroduced.

cjcliffe, 2015-05-28:
@dovandung I've added your patches to branch 'threading_fixes' and will be testing; thanks again!

cjcliffe, 2015-05-31:
closing for now, please feel free to open another issue if you discover anything :)

dovandung, 2015-05-31:
When run program with multiple VFO and delete VFO sometime program crash , when debug program I see the code to delete modulator seems not synchronize with the Thread using them so sometime the modulator is deleted but the Thread still using them . I modify function SDRPostThread::removeDemodulator(DemodulatorInstance *demod) as following it seems program run more stable . I test add multiple VFO , delete them and add another VFO it still work stable.void SDRPostThread::removeDemodulator(DemodulatorInstance *demod) { if (!demod) { return; }
 std::vector<DemodulatorInstance *>::iterator i;   //DovanDung modify  
 if (demodulators.size() != 0){ i = std::find(demodulators.begin(), demodulators.end(), demod);

 if (i != demodulators.end()) { demodulators.erase(i); } } }

cjcliffe, 2015-05-31:
I see the issue now and your fix improves the race condition but doesn't eliminate it.

I've replaced demodulators_add and demodulators_remove with busy_demod mutex lock so that demodulators won't be added/removed while data is being written.

The fix is committed to master; please pull the updates and let me know if that fixes the issue for you.

Thanks!

dovandung, 2015-06-01:
Hi Charles,The program is stable now .I have HackRF and SDRplay , If you make a framework I can add code for these device to CUBICSDR program. DovanDung

   
Reopened #81.—

cjcliffe, 2015-06-03:
If you have time to try replacing SDRThread code with equivalent using gr-osmosdr, SoapySDR or other I would gladly accept a working example and integrate.

Either of those would also work with RTL-SDR which would make it easier for me to work with.

Thanks for all your help!

closed 2015-06-03 by cjcliffe