cjcliffe/CubicSDR #89: Control speed of waterfall

Toontje, 2015-06-08:
I would love to see some way to control the waterfall speed. Currently the lower waterfall covers appx.12 seconds of traffic. I would like to see this configurable from very fast (2 seconds/screen) to very slow (2 minutes or more/screen). This is useful for scanning a band for unknown signals.

cjcliffe, 2015-06-08:
sounds good to me; I think I might aim for a configuration dialog in 0.1.4 and start filing some of the preferences like this one in there.   In the future I might also try an option for a non-linear waterfall where the lines slow and compress as they move to the bottom.

Toontje, 2015-07-02:
I think it would be great to have speed sliders on the right-hand side of the waterfall instead of having to set it in Preferences. That way you can adjust the speed on the fly when needed.

cjcliffe, 2015-08-14:
Waterfall and spectrum averaging can now be controlled from simple speed sliders to the right of the waterfall and spectrum.  Waterfall can go anywhere from 1 to 1024 lines per second and now continues to run while minimized/hidden.  Last speed settings will also be saved for next launch.

Cheers!

closed 2015-08-14 by cjcliffe