cjcliffe/CubicSDR #91: Buttons and TAB and SHIFT+TAB to skip to next/previous demodulator

cjcliffe, 2015-06-09:
[ ] TAB and SHIFT+TAB to select next/previous demodulator
[ ] Next/previous faded arrow buttons somewhere that appear on hover

Toontje, 2015-10-14:
I wanted to suggest this as well. I would like to be able to mute all demodulators except one. I would like to be able to easily switch the active demodulator.
Use case: Listen to all available frequencies for a near-by airport.
An automated signal detection that switches demodulator (scanner) would be a nice to have.

cjcliffe, 2016-02-11:
Hey @Toontje

In master branch:
- Tab and Shift-Tab now toggles the next/previous modem
- Solo mode will now mute everything except the active modem
- In Solo mode, if squelch is set and a squelch level is broken the modem will become active

Enjoy :)

closed 2016-02-11 by cjcliffe

Toontje, 2016-02-11:
Works like a breeze. Although i had to figure out what was happening, but i like the fact that the audio output device setting is linked to the demodulator. Opens a whole new box of use cases. ;-)
Great work again!!

cjcliffe, 2016-02-11:
@Toontje I think I'm enjoying the Solo+Tab feature combination a lot more than I imagined; just been dropping 10 or so SSB modems on active channels then comfortably surfing between them with Tab/Shift+Tab.

Toontje, 2016-02-12:
Hehehe. Good! I love to keep track of where the DX station is working contacts. It's an awesome feature that i think only SDR-Console has, but very very clumsy. This feature works much smoother in CubicSDR than what i know from SDR-Console where you have to set up multiple receiver sessions or something like that. Smooth working!! NICE!

cjcliffe, 2016-02-13:
@Toontje it revealed another necessity to me; I've added '[' and ']' keys that will allow you to nudge the active modem according to the current frequency snap setting.   

Real-world use case risk mitigation: I got a hand cramp while using my trackpad to keep making minor adjustments to all the solo'd ham chats I was toggling between ;)

Toontje, 2016-02-13:
Have to get my head around this. What did you just create? Nudge the active modem according to snap setting? What snap setting? You mean i can now finally do the 1 or 5kHz steps to switch between "channels"? Is that what you are talking about?

cjcliffe, 2016-02-13:
@Toontje When you right click the modem frequency digits they'll lock or "snap" the frequency to that digit (1's 10's 100's 1000's etc..) the nudge feature just moves the modem up or down at whatever the snap is set to.

Toontje, 2016-02-14:
Yeah, i remember that. Never used it extensively. WIll do now. Thanks! Can you lock on 5's? (nice-to-have, don't spend too much time on it)

cjcliffe, 2016-02-14:
@Toontje done, latest commit you can hold SHIFT key while using the [ nudge ] keys and it'll move at half the snap value; just set it one snap above what you'd like to move in steps of 5 :)

Toontje, 2016-02-15:
Awesome! Will try it out tonight!