cjcliffe/CubicSDR #92: Manual Gain controls

cjcliffe, 2015-06-09:
- Allow enable/disable of the built-in automatic AGC states
- Add a new slider control somewhere that appears for manual gain control mode.

Gussi, 2015-06-11:
Manual gain control would be fantastic. :+1:

Toontje, 2015-07-02:
+1 although i think the automatic gain works great. Not a high priority for me.

cjcliffe, 2015-08-27:
Going to bump this to 0.1.5 so that I can get the 0.1.4 build out this week; also might do it in tandem with the gr-osmosdr stuff since the gain controls may be different across devices.

dc1rdb, 2015-09-21:
With SoapySDR integration progressing so well, maybe this could be changed into "Provide submenu for frontend-specific controls" like:
- Manual gain control for LNA, mixer, IF, as applicable
- AGC on/off
- PreAmp on/off
- Antenna power on/off
- etc.

cjcliffe, 2015-10-26:
Almost ready here; should have a working commit tomorrow.

cjcliffe, 2015-10-27:
Manual Gain control now committed to soapysdr-support branch; requires latest SoapySDRPlay and SoapyHackRF drivers for proper gain control.  SoapyRTLSDR should be already good to go.

Just disable AGC from settings menu and the manual gain controls will appear; enjoy :-)

closed 2015-10-30 by cjcliffe