cjcliffe/CubicSDR #98: Display only refreshes on mouse movement in El Capitan

mikelagasse, 2015-06-24:
In OS X El Capitan DP2, The entire contents of the display in CubicSDR do not refresh unless the mouse is currently moving.

cjcliffe, 2015-06-24:
Wow that's an oddball -- The entire display refreshes on mouse move (like all views) or just the view the mouse is currently over?

mikelagasse, 2015-06-24:
I'm not sure the distinction, but nothing within the CubicSDR window (ex: graphs, waterfall) refreshes unless the mouse is actively moving. Audio does continue to play back. As long as you constantly wiggle the mouse everything updates smoothly.

cjcliffe, 2015-06-25:
The waterfall, scope, etc are all separate wx panels at the moment so it's strange -- I'm guessing there's some missed refresh event at the AppFrame that's now interpreted as a non-update.

I will investigate, thanks.

cjcliffe, 2015-08-10:
@mikelagasse if you (or someone) is able to test the latest commits to master to see if this is resolved in El Capitan I'd appreciate it.

thanks!

alienshop, 2015-09-14:
Hi, I can confirm that its now working correctly in El Capitan with release 0.1.4. Thanks!

cjcliffe, 2015-09-14:
perfect, cheers!

closed 2015-09-14 by cjcliffe