cjcliffe/CubicSDR #102: Program startup with default preferences file

Toontje, 2015-07-02:
I use different preferences files for when i am using CubicSDR as SDR program or when i'm using it as panadapter. Would it be possible to start the program with a default preferences file that can be set up for most common use?

cjcliffe, 2015-07-17:
Would a simple command line parameter that takes a name for the configuration be enough?  

It would be simple to just redirect the configuration file to something like cubicsdr-[configname].xml where [configname] is supplied via command line parameters.

Toontje, 2015-07-18:
That would be great. And if no command line parameters are given it loads a
default XML file.

cjcliffe, 2015-07-21:
Latest master will now accept -c [name] or --config [name] to set the config-[name].xml as the active config; default config will be copied to the new one if no file exists.

Thanks!

closed 2015-07-21 by cjcliffe