cjcliffe/CubicSDR #104: Audio recording, levels, etc.

redbobcat, 2015-07-03:
Just now tried about 3h direct sampling recv on HF. I think, it is needs a lot:
Waterfall speed control - hard to understand modulation on such slow waterfall, mostly CW looks like constant carrier now.
Y axis of panadapter needs levels labelling (like in gqrx and others). Auto range is great, but manual will be good to (amp testing, noise levels investigation). And averaging control.
Spectrum/scope switch will be great, sometimes spectrum is very useful.

cjcliffe, 2015-08-14:
In master branch the spectrum can now be controlled via a speed control slider on the right.

cjcliffe, 2015-08-26:
- Main I/Q spectrum and audio spectrum now feature peak and floor decibels display
- Waterfall speed control can now be set between 1 and 1024 lines per second
- Waterfall now continues while in background or minimized
- A new spectrum visual is available by dragging left on the scope area, more visuals may be added here :)

closed 2015-08-26 by cjcliffe