cjcliffe/CubicSDR #105: Bug in freq jump

redbobcat, 2015-07-06:
After hiting space, in pop-up window, if you do not use dot (XX  <s>.X</s>), jumps to 1.2Mhz, if dot placed, (XX.X) all works good.

cjcliffe, 2015-07-06:
It's not documented at the moment, but if you don't supply a decimal place it assumes the frequency is in Hz -- you can also apply Ghz as a suffix for stuff like 1.090Ghz.   

Though, on second thought it might be wise to adjust it so that any frequency specified without a decimal that's less than something like 1500hz (minimum allowed demod bw) would logically translate to Mhz.

redbobcat, 2015-07-06:
Would be nice, offten thinks in MHz, and 1Ghz logicaly is 1000Mhz, and any other thinks is MHz.
How else can I help without programming skils?

cjcliffe, 2015-07-06:
I appreciate the help finding quirks like this -- you can always try tweaking some variables somewhere or changing random things on your own to see how it works to get started or you might be more interested once some basic LUA integration is available for tinkering with things like scripts to decode digital demodulator stream outputs. 

Thanks!

cjcliffe, 2015-07-07:
direct input will now assume Mhz when the value is <= 3000.  If you want to use 3000hz or less you can now use the hz suffix.

now in master and will be in next release, cheers!

closed 2015-07-07 by cjcliffe