cjcliffe/CubicSDR #119: OS X config: High resolution and graphics switching

jacobseptember, 2015-07-26:
Two simple fixes that can be accomplished by adding keys to Info.plist:

	<key>NSHighResolutionCapable</key>
	<true/>
	<key>NSSupportsAutomaticGraphicsSwitching</key>
	<true/>

The high-resolution bit allows fonts/ dialog boxes, etc to display at Retina resolutions instead of being scaled up forcefully (which makes them pixellated). The graphics switching means the application won't force the use of discrete graphics instead of integrated.

andresv, 2016-02-12:
I am glad I found this ticket. It was awful without this fix. You do not want merge this sooner?

cjcliffe, 2016-02-13:
@andresv I've honestly just not tackled this as I don't have any way to test it but reviewing again I see now it's a really minor change -- I'll find out what's required to add it to the Info.plist file and post here for testing unless you've got a known working patch you'd like to submit.

jacobseptember, 2016-02-13:
Adding those keys should break absolutely nothing. At least the high-res-capable one. If you really want to force discrete graphics I guess you can leave the other one out, but it works fine on my laptop.

cjcliffe, 2016-02-13:
@jacobseptember just taking a look now actually; I assumed there was some magic to the Cmake OSX plist generator but it's literally a copy-paste job to include it in the project.  I'll have an updated binary for testing to confirm shortly.

cjcliffe, 2016-02-13:
@jacobseptember @andresv Here's an updated version 
[CubicSDR-0.1.25-Darwin.dmg.zip](https://github.com/cjcliffe/CubicSDR/files/128968/CubicSDR-0.1.25-Darwin.dmg.zip) with the options compiled in.  

I've committed a plist .in file and relevant CMakeLists entries for generating the patched file automatically for future releases.

Apologies for not patching it up sooner.   I'd have liked to evaluate it and make any tweaks that it might need for the OpenGL generated fonts and widgets, etc. but if this helps for now and is usable then that's at least an improvement.

andresv, 2016-02-13:
I currently do not have SDR connected, however this is how it look likes without one:
[image]

andresv, 2016-02-13:
I noticed that  `SoapySDR Device Options` table is still pixelated:
[image]

jacobseptember, 2016-02-13:
Looks ​nice on my end! As Andres mentioned, system-generated fonts (window
title, hover text, "open file" dialog box) all look perfect.
Program-generated fonts (everything in-window, basically) are still
pixellated. This has always been the case when I modified my own Info.plist.

Thanks!

cjcliffe, 2016-02-14:
@andresv @jacobseptember looks reasonable to me for now; not sure why some wx stuff would still be pixelated; possibly a bug within wxWidgets -- I'm going to bump this issue to after the next release and look into improving the resolution of a few of the OpenGL UI elements as well.