cjcliffe/CubicSDR #123: Filter bandwidth only set when hovered over

Toontje, 2015-08-01:
Check the video here: https://www.dropbox.com/s/ryy7tjz6w29wvlq/CubikSDR%20Bug.mov?dl=0
Some operations are on my second screen, but i will explain:

At 0:09 you see that i launch CubicSDR, move it to my second screen and resize.
At 0:22 i open the Settings file. Note that filter bandwidth is set to 200.000. In the settings file it is set to 6000.
You see that the center frequency changes but the filter bandwidth not.
At 0:42 i hover over the filter and then the filter bandwidth is set to the correct value.

Bug?

cjcliffe, 2015-08-01:
Not really a bug; what happens is that it loads and activates the demodulator but doesn't select it by default since there's the possibility of multiple demodulators loading; 200.000 is still just the unchanged default before you select it.

What you're seeing is when there's nothing selected and you hover over a demodulator the default action is to select it; otherwise it just gives you a scope preview on hover if it's not the selected one.

Though I can make it so that if there's only a single instance in the file it would select it by default; that might be more intuitive for when you're loading a session file containing a single frequency.

Toontje, 2015-08-01:
Yeah, that would be more clear, I think.

cjcliffe, 2015-08-17:
If there's a single instance in the session file it'll now be selected and centered in view on load.

Cheers!

closed 2015-08-17 by cjcliffe