cjcliffe/CubicSDR #128: "dat-gui" style dropdown configuration panel

cjcliffe, 2015-08-10:
Create a simple "dat-gui" style (simple stack of rows with various basic controls) that can be shown/hidden on each relevant panel to provide access to advanced context-relevant controls without cluttering the interface with rarely used buttons..

cjcliffe, 2019-01-04:
Stuffing more things in separate wxWidgets panels is a lost cause I believe; I'm leaning towards replacing wxWidgets with Nuklear -- which makes this issue irrelevant.

closed 2019-01-04 by cjcliffe

vsonnier, 2019-01-10:
Hi Charles ! First, I wish you a very Happy New Year !
I saw that you plan to continue to work on CubicSDR and I'll be happy to continue helping :)
Thanks to you I discovered both https://github.com/vurtun/nuklear and https://github.com/ocornut/imgui.
Looking at the two, I have the feeling that Nuklear is "cheap" compared to ImGui (judging by the demos and screenshots alone) with its C-isms and having less widgets, and apparent less support (help by either dev or general users) etc. compared to ImGui.

I think that Nuklear would be more oriented for "basic" GUI for the light, embedded world than to create super-sexy GUI like ImGui does for "normal" PCs.

Here is my 2 cents...
Regards, 
Vincent