cjcliffe/CubicSDR #132: build error

k9wkj, 2015-08-16:
~/CubicSDR/src/demod/DemodulatorThread.h:53:5: error: ‘freqdem’ does not name a type
     freqdem demodFM;
     ^
CMakeFiles/CubicSDR.dir/build.make:54: recipe for target 'CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o' failed
make[2]: *** [CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o] Error 1
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/CubicSDR.dir/all' failed
make[1]: *** [CMakeFiles/CubicSDR.dir/all] Error 2
Makefile:75: recipe for target 'all' failed
make: *** [all] Error 2

 on debian stretch

cjcliffe, 2015-08-17:
Ahoy, did you follow the instructions from https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux ?  I think there's  a slightly older version of liquid-dsp that doesn't have freqdem separate from freqmodem; make sure you've got the latest one.

k9wkj, 2015-08-17:
ohh  yes
now i remember!!
i had givin up on building it a couple months ago because i was running Sid
and i couldnt get half the deps to build
so i ran the package out of the repos  and that was fine
but youve changed a bunch of stuff  and i thought i would build it
hahahaha
one of the reasons i was looking into it again
is how broken rtl_tcp is     utterly useless
and i ran across a fella here on github that wrote his own tcp transport
protocol
and it makes a remote dongle appear as a local one
https://github.com/texane/librtlsdr/tree/rpc
and it seems to work out fairly well
but i wanted to mess with the starting bitrate in Cubic  and see if it
could start at 250k or so
as it really is struggling for some reason at even 1mbit using the AM
demodulator
gqrx seems to be pretty responsive using the remote lib

ill keep messing when i get more time
thanks! for the great work

Wayne

cul de k9wkj

cjcliffe, 2015-08-27:
Hope it works out for you; 250k mode is experimental but seems usable in my tests for local receiving -- I'd be interested to know if that remote driver is stable with it.

feel free to post another issue if you're still having build problems.

Cheers!

closed 2015-08-27 by cjcliffe