cjcliffe/CubicSDR #139: Possible issue with window management on Mac OSX

Toontje, 2015-09-01:
I normally work with two screens where i have CubicSDR on my second screen. CubicSDR/OSX (not sure who) remembers on which screen CubicSDR was last run.
Normal behaviour is that, when i disconnect my second screen, all windows are gathered on my primary screen. However, CubicSDR stays on the second screen coordinates and because the window cannot be accessed it also cannot be moved to the primary screen.
Not sure if this is a CubicSDR issue or an OSX setting.

cjcliffe, 2015-09-04:
Most likely an issue with wxWidgets; I haven't compiled anything newer than 3.0.2 for OSX yet and I'm sure there's been fixes for Yosemite (as well as full-screen support) added since then.   I'm going to start doing more test builds after each issue so that more people have access to try out features before final release -- 0.1.4 appears to be buggy and if I'd had a few more eyes on it I'm sure issues would have been resolved before then.

Thanks for all your help so far!  Will get back to you soon on this.

cjcliffe, 2015-09-04:
@Toontje please give https://github.com/cjcliffe/CubicSDR/releases/tag/0.1.5-issue139 a try and let me know if that solves anything

Toontje, 2015-09-04:
Trying it out now. Half of the screen is now on the primary display and half on the secondary. There is no way to move the complete window to the primary screen if there is no secondary screen present.
Default behaviour of OSX applications is to move 100% to the primary screen if the secondary screen is not present. This is still not happening.

cjcliffe, 2015-09-04:
There must be some event or condition I'm not handling then; I would assume this was a window manager's job but I'll see what I can do to detect screen changes like that.

Toontje, 2015-09-05:
I am now working on my two screens and this is what happens:
- When i have CubicSDR on my second screen and i disconnect the screen CubicSDR gets moved correctly to the primary screen.
- When i close CubicSDR on my second screen, disconnect the second screen and start CubicSDR, CubicSDR shows only the lower part of the window on the upper part of the screen. See https://www.dropbox.com/s/xbxit17n8fk6uob/CubicSDR.png?dl=0

cjcliffe, 2015-09-09:
Haven't had a chance to fix this up yet; I had something that sort of worked but it triggered sometimes when it shouldn't.

Not sure if this helps but I had a window stuck off the top and right of the screen and was monkeying with it and found a trick.  If you get the up/down or left/right resize arrow on a widow edge just click and instead of making the usual resize movement, drag along the edge (drag left-right on the top/bottom of the window or drag up/down on the sides of it)  -- OSX will pick up the window and let you move it instead of resize.

Toontje, 2015-09-13:
Yeah, i found a hint that worked for me, use Spaces to select the app, you won't see it but the menu will be present, then from the menu select Window, Zoom. Then the app shows up.
BTW, CubicSDR is not the only app with this window behaviour. I wouldn't make it a high priority.

cjcliffe, 2016-01-12:
Going to close this one and assume wxWidgets will improve this in the next version (looks like they'll support the "full screen" modes as well) as I've never been able to improve it so far without creating more window manager problems and fragmenting platforms :)

closed 2016-01-12 by cjcliffe