cjcliffe/CubicSDR #141: Add one more keyboard shortcut

Toontje, 2015-09-05:
The ? which shows all available keyboard shortcuts.

Toontje, 2015-10-22:
Found something i have not seen before. How many of those "easter eggs" am i still missing? ;-)

https://www.dropbox.com/s/9dvshlhlbjghljg/new.mov?dl=0

cjcliffe, 2016-02-08:
I've started working on these in the keyboard_updates branch; some of it working already.

cjcliffe, 2016-02-09:
@Toontje I tried doing the '?' showing a help popup but it was rather clunky and hard to reference to what you were currently doing.

During the process I had the idea to do something a bit more helpful for new users and make the hover tips from the status line show as a standard ToolTip popup beside the mouse when you hover the various parts of the application.  

I've also added first Settings menu option as "Enable Hover Tips" so that once you've learned what you need you can turn them off and by that time have probably noticed they're the same tips as in the status line anyways.  This setting will be saved.

Hopefully this helps new users to get going immediately to understand the features and experienced users will likely find it annoying after a few minutes and just turn it off.

closed 2016-02-09 by cjcliffe

Toontje, 2016-02-09:
I thik this actually works fine. And yes, this is for starters only. Once you know how to use the program you would want to switch it off. So good solution. Thanks!