cjcliffe/CubicSDR #145: Suport for tunning over TCP/IP

ewpereira, 2015-09-13:
Hello Charles,

First I would like to thank and congratulate you for all your work on making CubicSDR avalilable. It is simply very nice.

I am developing some software for operating cubesats and for the ones that have analog transponders (like the FUNCube-1), I would like to monitor the telemetry in the beacon frequency as well as the transponder passband using separate audio output devices. CubicSDR allows me to do just that, but it would be great to be able to have automatic tuning to compensate for the doppler shift.

One possible way to have automatic tuning would be to have a TCP socket in CubicSDR for remote control. A protocol like JSON could be used to maintain OS idependence. An external application (Keplerian Propagator) could then be used to remote control the frequency of each receiver in CubicSDR.

If you think this could be something useful to be added to CubicSDR, I could try to implement it (time permiting).

73, Edson PY2SDR
http://wabicafe.com.br/py2sdr

cjcliffe, 2015-09-13:
Thanks for your support!

Not sure when I'll have a chance to start on it but I'd like to allow support for simple add-ons like this through LUA scripting which is planned for the future.  If you'd like to try something in the meantime the DataTree might be suitable for simple XML exchange and wxWidgets has a socket client/server network support and both are already built-in.  
Also If there's a remote control specification available that's already used by other applications it might be pertinent to look at implementing one of those options first.

Cheers,
-CJ

ewpereira, 2015-09-13:
Hello Charles,

One possibility is to use the same control interface as the one used in
SDRSharp Net Remote plugin (
http://eartoearoak.com/software/sdrsharp-net-remote). It uses JSON for the
line protocol.

There are many SDR software that emulate some of the protocols used for
hardware (physical radios). But I think it would be easier (and better) to
use a network protocol like JSON since it is XML based and rather simple.

Cheers,

Edson

---
- We humans have the capability to do amazing things if we work together.
- Nós seres humanos temos a capacidade de fazer coisas incríveis se
trabalharmos juntos.