cjcliffe/CubicSDR #149: Remove unnecessary null pointer checks

elfring, 2015-09-25:
[Extra null pointer checks are not needed in functions](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for NULL before delete p?") like the following.
* [DataElement](https://github.com/cjcliffe/CubicSDR/blob/dfd02c1b1213d4250902872b0ede5ddae89f5c56/src/util/DataTree.cpp#L39 "Destructor for DataElement")
* [RtAudio::openRtApi](https://github.com/cjcliffe/CubicSDR/blob/dfd02c1b1213d4250902872b0ede5ddae89f5c56/external/rtaudio/RtAudio.cpp#L119 "RtAudio::openRtApi function")
* [WaterfallPanel::setup](https://github.com/cjcliffe/CubicSDR/blob/9852967551904507599e6bc95a73e28a84bb8495/src/panel/WaterfallPanel.cpp#L10 "WaterfallPanel::setup function")
* [WaterfallPanel::step](https://github.com/cjcliffe/CubicSDR/blob/9852967551904507599e6bc95a73e28a84bb8495/src/panel/WaterfallPanel.cpp#L52 "WaterfallPanel::step function")

cjcliffe, 2015-09-25:
@elfring thanks there's some handy info there; I see you're promoting proper C++ use for a lot of repositories -- are you on a mission with a set of canned responses or are these posts coming from an automated static analysis tool you're rolling through GitHub? :)

elfring, 2015-09-26:
A big part of my mission is to improve various free software.

How do you think about to improve [static source code analysis](https://en.wikipedia.org/wiki/List_of_tools_for_static_code_analysis#C.2FC.2B.2B "Tools for static code analysis") also for your software?

closed 2015-11-13 by cjcliffe

elfring, 2015-11-13:
There are [still a few update candidates left over](https://github.com/cjcliffe/CubicSDR/blob/7a01c436d64b6fcf36e6cf69ed21ccbe977d64eb/src/util/DataTree.cpp#L46 "Update candidate: DataElement::data_init()") after [your software change](https://github.com/cjcliffe/CubicSDR/commit/538092d9a03a6ce3b504ff5ab7f9a120f8ee803e "Deletion of a null pointer check").

cjcliffe, 2015-11-13:
@elfring thanks I've reviewed and that's all I'll be updating for now; the other cases aren't relevant at the moment.