cjcliffe/CubicSDR #151: Airspy not working (branch soapysdr-support)

antigraviton, 2015-09-28:
Dear Charles,

I've tried running CubicSDR with the airspy. Initially I get an error (from Soapy), about an unsupported samplerate.

```
 ~/dev/rx/CubicSDR/build/x64 $ sudo ./CubicSDR 
Loading:: configuration file '/home/dtegeler/.CubicSDR/config.xml'
Loaded PPM for device 'Generic RTL2832U OEM :: 00000001' at 57ppm
Loaded I/Q Swap for device 'Generic RTL2832U OEM :: 00000001' as not swapped
Loaded Direct Sampling Mode for device 'Generic RTL2832U OEM :: 00000001':  off
Loaded offset for device 'Generic RTL2832U OEM :: 00000001' at 0Hz
SoapySDR init..
        API Version: v0.3.0-g9671ca26
        ABI Version: v0.3-0
        Install root: /usr/local
        Module found: /usr/local/lib/SoapySDR/modules/libairspySupport.so
        Module found: /usr/local/lib/SoapySDR/modules/libhackrfSupport.so
        Module found: /usr/local/lib/SoapySDR/modules/librfspaceSupport.so
        Module found: /usr/local/lib/SoapySDR/modules/librtlSupport.so
        Loading modules... done
        Available factories...airspy, hackrf, null, rfspace, rtl
Found device 0
  airspy = 0
  driver = airspy
  label = AirSpy AIRSPY
Make device 0
Using AIRSPY with firmware AirSpy NOS v1.0.0-rc6-0-g035ff81 2015-07-14 

SDR post-processing thread started..
Spectrum visual data thread started.
SDR thread initializing..
Using device #0
Spectrum visual data thread started.
Using AIRSPY with firmware AirSpy NOS v1.0.0-rc6-0-g035ff81 2015-07-14 
terminate called after throwing an instance of 'std::runtime_error'
  what():  airspy_set_samplerate(2.4e+06) has failed (-9999) AIRSPY_ERROR_OTHER
```

Hardcoding the sample rate in `sdr/SoapySDRThread.cpp` line 177, i.e. replacing `sampleRate.load()` with `2.5e6` solves the initial startup problems, however it looks like there's something else wrong.
Also tried changing the default sample rate `DEFAULT_SAMPLE_RATE` to 2500000, with the same result.
Difficult to describe aside from 2 weakish signals that don't move when changing frequencies, see below.

[image]

How can I help?

Kind regards (and many compliments for your work),

Derk

antigraviton, 2015-09-28:
update: issues fixed with:
* Firmware Version: AirSpy NOS v1.0.0-rc5-0-g648c14f 2015-05-20
* DEFAULT_SAMPLE_RATE set to 2.5e6

The firmware issue will be solved (eventually) in SoapyOsmo I expect (bit packing added to firmware)

Absence of gain controls (we have 3 for the airspy), makes it unusable unfortunately:
[-v vga_gain]: Set VGA/IF gain, 0-15 (default 5)
[-m mixer_gain]: Set Mixer gain, 0-15 (default 5)
[-l lna_gain]: Set LNA gain, 0-14 (default 1)

Setting other defaults in SoapyOsmo (gr-osmosdr/lib/airspy/airspy_source_c.cc) did the trick! it works, much better than gqrx at 10MSPS on under powered hardware.

Fantastic.

cjcliffe, 2015-09-28:
Aye, I should have some basic gain controls in this week for testing, just working out some other things and then I'll have a test dialog that can adjust whatever gain parameters SoapySDR exposes (assuming the OsmoSDR module supports it).  

I'll update this issue when it's ready for testing -- thanks for your help!  Ultimately I think we'll want a native standalone SoapyAirSpy to replace the OsmosSDR stuff as there's only a few devices left that we don't already cover with standalone modules and it removes a nice pile of dependencies in most cases.

cjcliffe, 2015-10-18:
@antigraviton haven't got the gain support in yet; but I've pushed a pile of optimizations to the soapysdr-support branch -- let me know if that improves performance for you on the AirSpy.

closed 2015-11-10 by cjcliffe