cjcliffe/CubicSDR #152: Manual Visual Scale control

Toontje, 2015-09-30:
Hi Charles!

I would like to suggest a way to manual control the visual scale. Automatic as it is now works fine if you are looking at a band where all signals are approx at the same level. But when you are looking at a band where part of the signals are weak and some strong signals are present you cannot see the weak signals because Cubic adjusts the visual scale to the strong signals. This is specially important when using wide-band devices like the SDRplay.
Example: When i look at an overview of the 20m amateur band i cannot see any signals unless i exclude the 19 and 22m broadcast band from the display by either changing the input bandwidth or using zoom.

cjcliffe, 2015-09-30:
I've been wanting to do this for awhile, I'd like to make it so you can just hold ALT and drag on the spectrum to "pull up" or "push down" the spectrum -- does that sound like a good method?  

I'm thinking as soon as you zoom out or move enough it'll just reset on it's own or double click or escape to reset it.

Toontje, 2015-10-01:
Yes, that would be good.

UbhaSecurity, 2015-10-01:
Agree.

Same issue when I add lna in my chain.

Like to use alt and up down.

cjcliffe, 2015-10-25:
I've added support for this in soapysdr-support branch, after some tinkering I set it as right-drag up/down to adjust scale on the spectrum.  Single right click without dragging or zooming significantly will reset it.

Toontje, 2015-10-25:
Works a expected except in the upper right panel(s).

cjcliffe, 2015-10-25:
I'm probably removing A/Z in favour of being able to push 'A' for AM, there's already scroll wheels and arrow keys for zooming/panning.  I could easily add it to SHIFT-UP and SHIFT-DOWN as they're currently not assigned and kinda makes logical sense to "shift it up/down"

cjcliffe, 2015-10-25:
Latest commit now has SHIFT + UP / DOWN keys for adjusting visual scale; more fine-grained than dragging it.

Toontje, 2015-10-26:
Great stuff!! Working as expected!! Good!!

closed 2015-10-30 by cjcliffe