cjcliffe/CubicSDR #154: Bug: [xcb] Unknown request in queue while dequeuing

UbhaSecurity, 2015-10-05:
ulf@hack:~/src/new/CubicSDR/build/x64$ ./CubicSDR 
Loading:: configuration file '/home/ulf/.CubicSDR/config.xml'
Loaded PPM for device '' at 0ppm
Loaded I/Q Swap for device '' as not swapped
Loaded Direct Sampling Mode for device '':  off
Loaded offset for device '' at 0Hz
Loaded PPM for device 'Generic RTL2832U OEM :: 00000001' at 0ppm
Loaded I/Q Swap for device 'Generic RTL2832U OEM :: 00000001' as not swapped
Loaded Direct Sampling Mode for device 'Generic RTL2832U OEM :: 00000001':  off
Loaded offset for device 'Generic RTL2832U OEM :: 00000001' at 0Hz
Loaded PPM for device 'Realtek RTL2838UHIDIR SN: 00000001' at 0ppm
Loaded I/Q Swap for device 'Realtek RTL2838UHIDIR SN: 00000001' as not swapped
Loaded Direct Sampling Mode for device 'Realtek RTL2838UHIDIR SN: 00000001':  off
Loaded offset for device 'Realtek RTL2838UHIDIR SN: 00000001' at 0Hz
Spectrum visual data thread started.
Spectrum visual data thread started.
SDR post-processing thread started..
FFT visual data thread started.

Audio Device #0 PulseAudio
	Default Output? Yes
	Default Input? Yes
	Input channels: 2
	Output channels: 2
	Duplex channels: 2
	Native formats:
		16-bit signed integer.
		32-bit signed integer.
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
		8000hz
		16000hz
		22050hz
		32000hz
		44100hz
		48000hz
		96000hz

Available vertical sync SwapInterval functions: 
	glxSwapIntervalEXT: No
	DRI2SwapInterval: No
	glxSwapIntervalMESA: No
	glxSwapIntervalSGI: No
No vertical sync swap interval functions available.
Loaded font 'Bitstream Vera Sans Mono' from 'vera_sans_mono16.fnt', parsed 167 characters.
Loaded font 'Bitstream Vera Sans Mono' from 'vera_sans_mono12.fnt', parsed 167 characters.
Loaded font 'Bitstream Vera Sans Mono' from 'vera_sans_mono24.fnt', parsed 167 characters.
SDR enumerator starting.
[xcb] Unknown request in queue while dequeuing
[xcb] Most likely this is a multi-threaded client and XInitThreads has not been called
[xcb] Aborting, sorry about that.
CubicSDR: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
CubicSDR: ../../src/xcb_io.c:179: dequeue_pending_request: Assertion `!xcb_xlib_unknown_req_in_deq' failed.
Aborted

Latest git soapysdr/cubic

cjcliffe, 2015-10-05:
Yeah I'm getting a similar error right there on OSX about a CoreAnimation in queue that was deleted before it finished -- but OSX just plows through it anyways.. 

I think I've wired up something funky in the new selector dialog init -- I'll take a look shortly and debug that error here.

cjcliffe, 2015-10-05:
yeah according to the interwebs I'm initializing a UI context from the wrong thread.. sounds like I'll just make appFrame launch the selector instead! :)  fix coming shortly..

cjcliffe, 2015-10-06:
@UbhaSecurity latest push should do the trick, was jumping around between raspberry pi remote devices and locals with the new dialog last night without much trouble :)

Remotes will eventually save, but I need to decouple it from the local devices and allow it to build the device list asynchronously as they finish querying.  That way you won't have to wait for all the remotes on startup and just jump to a device when it's ready.

cjcliffe, 2015-10-09:
@UbhaSecurity wasn't fixed in latest commits when I tried in linux here; should be actually fixed now.

cjcliffe, 2015-10-14:
Can't reproduce this one anymore here.

Koatman, 2015-12-15:
Hello,
I’ve just found out about this program and got the same error on a current Arch Linux install after compiling from git

$ CubicSDR 
Spectrum visual data thread started.
SDR post-processing thread started..
Spectrum visual data thread started.
FFT visual data thread started.

Audio Device #0 PulseAudio
	Default Output? Yes
	Default Input? Yes
	Input channels: 2
	Output channels: 2
	Duplex channels: 2
	Native formats:
		16-bit signed integer.
		32-bit signed integer.
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
		8000hz
		16000hz
		22050hz
		32000hz
		44100hz
		48000hz
		96000hz

Available vertical sync SwapInterval functions: 
	glxSwapIntervalEXT: No
	DRI2SwapInterval: No
	glxSwapIntervalMESA: No
	glxSwapIntervalSGI: No
No vertical sync swap interval functions available.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/vera_sans_mono16.fnt', parsed 167 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/vera_sans_mono12.fnt', parsed 167 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/vera_sans_mono18.fnt', parsed 167 characters.
SDR enumerator starting.
SoapySDR init..
	API Version: v0.4.0-g7c716a29
	ABI Version: v0.4-2
	Install root: /usr/local
	Loading modules... 
	Available factories...nullJust 'null' factory found.
[xcb] Unknown request in queue while dequeuing
[xcb] Most likely this is a multi-threaded client and XInitThreads has not been called
[xcb] Aborting, sorry about that.
CubicSDR: xcb_io.c:179: dequeue_pending_request: Assertion `!xcb_xlib_unknown_req_in_deq' failed.
CubicSDR: Fatal IO error 0 (Success) on X server :0.0.
Abgebrochen (Speicherabzug geschrieben)

cjcliffe, 2015-12-15:
@Koatman I see "Available factories...nullJust 'null' factory found." -- I think this is related to the new Alert dialog that's supposed to warn you when there's no modules found.  I'm betting if you install SoapyRTLSDR, SoapyRemote or whatever Soapy module for your device it will bypass that error..

Will patch that up -- I likely called the alert dialog from the wrong thread :)

Koatman, 2015-12-15:
Correct!  I installed the BladeRF soapy module.  The program runs now.  Thanks!

cjcliffe, 2015-12-16:
@Koatman excellent, I'll add an issue to fix the new crash you encountered as well.  cheers!

closed 2015-12-16 by cjcliffe