cjcliffe/CubicSDR #157: CubicSDR-0.1.8-win32.exe 0xc000007b

vsboost, 2015-10-11:
Just tried installing CubicSDR-0.1.8-win32.exe and get.

The application was unable to start correctly (0xc000007b). Click OK to close the application.

Am i missing something else that needs to be installed?

cjcliffe, 2015-10-11:
Apologies as I realize the individual release links don't show the history of the previous; I've updated the description with the relevant links; you'll need to build at least SoapySDR and then the driver of your choice and alternately SoapyRemote.  I believe SoapySDR binaries will be available soon.

vsboost, 2015-10-11:
I installed the PothosSDR binary from http://downloads.myriadrf.org/builds/PothosSDR/ and tested the hackrf with SoapySDRUtil --probe="driver=hackrf" which worked, so i take it SoapySDR is different from PothosSDR.

cjcliffe, 2015-10-11:
Hmm, it may only work with the 32-bit version; did you grab the 64 bit version?   I wasn't able to get the 64-bit CubicSDR/SoapySDR binaries working here yet.

I may have to try installing it myself to see.

vsboost, 2015-10-11:
Yea i grabbed the newest one 	PothosSDR-2015.10.05-vc12-x64.exe

cjcliffe, 2015-10-11:
Yeah I haven't been able to make my 64-bit builds work here, I'll try installing that and see if compiling against it helps.  If it works I'll add the 64-bit build.

vsboost, 2015-10-14:
Any luck with this?

cjcliffe, 2015-10-14:
Not yet; trying to install VS2015CE but having no luck, installer keeps crashing -- will update when I have something ready.

vsboost, 2015-10-14:
Thanks Charles

cjcliffe, 2015-10-19:
@vsboost after resetting windows 10 on my laptop partition I was able to finally install VS2015 and produce a build that seems to work here.  It's available now at https://github.com/cjcliffe/CubicSDR/releases/tag/v0.1.11 -- let me know how that works for you.

vsboost, 2015-10-19:
Tried it but crashes on start up.

[image]

vsboost, 2015-10-19:
If I run the exe in compatibility mode for windows 8 it works.

cjcliffe, 2015-10-29:
I've made some additional fixes that were major problems for windows builds; I'm going to try out the new SoapySDR loader features and try to produce an installer with a few modules bundled -- I'll update this issue once it's ready.

vsboost, 2015-10-29:
Thanks Charles,  and thank you for all your hard work it is definitely appreciated, i notice some forums are already talking up CubicSDR and i believe that one day it will be the go to program for SDR enthusiasts.

Cheers

cjcliffe, 2015-10-29:
@vsboost thanks for the encouragement -- still hard to believe I picked up my first RTL-SDR dongle just over a year ago and pushed my first working commit one year ago yesterday..  I guess that makes this it's first anniversary :)

vsboost, 2015-10-29:
Truly an achievement, i just wish i knew how to program as i would have loved to contribute code.

cjcliffe, 2015-10-30:
@vsboost try out https://github.com/cjcliffe/CubicSDR/releases/tag/v0.1.15-alpha and let me know how that works for you -- it should be completely self-contained but will also use any compatible local SoapySDR modules as well.

vsboost, 2015-10-30:
Hey Charles got some strange results, tried it on my Windows 8.1 work machine and picked up my E4000 dongle straight away (no pothossdr installed).  Tried it on my Surface Pro 2 with Windows 10 and wont pick up my HackRF or R820T2 i installed PothosSDR and it then picked them both up.

Ill do some more testing and see if i can narrow it down some more.

vsboost, 2015-10-30:
Disregard some of the above, i rebooted my win 10 machine and it now picks up my rtl dongles it just wont pick up the hackrf unless i have pothossdr installed.

cjcliffe, 2015-10-30:
@vsboost are you able to do a bit of debugging and move the DLL's from the /Program Files/PothosSDR/bin/ directory somewhere then move anything that looks relevant to HackRF back in one at a time until it works to see what's missing?

Cheers!

vsboost, 2015-10-30:
Yep the file pthreadVC2.dll once moved out of C:\Program Files\PothosSDR\bin will not detect the HackRF, if that file is moved into the CubicSDR directory it picks up the HackRF no problem.

cjcliffe, 2015-10-30:
awesome thanks; I'll add that in to the next bundle test

cjcliffe, 2015-11-10:
SoapySDR Support has now been merged down to master and there are usable instructions for building for Windows in the Wiki now. 

Windows 32 should also be able to be built (I've had it working previously) by replacing instances of 64/x64 with 32/x86 where appropriate in the instructions.

Going to close this one up for now, cheers!

closed 2015-11-10 by cjcliffe