cjcliffe/CubicSDR #158: Does not seem to want to install on Ubuntu 12.04

ishan-marikar, 2015-10-11:
Hey there, I tried installing CubicSDR and came across some dependency issues, so I installed the relevant dependencies.
```
sudo add-apt-repository -y ppa:kamalmostafa/wxwidgets3.0 && sudo apt-add-repository ppa:gqrx/releases -y  && sudo apt-get update && sudo apt-get install librtlsdr librtlsdr-dev libwxgtk3.0-0 libwxgtk3.0-dev libfftw3-3 libfftw3-dev   

```

.. but it keeps telling me that they aren't installed. Does CubicSDR not work on 12.04 or am I just dumb? :c

cjcliffe, 2015-10-11:
Not sure, it installed and ran with RTL-SDR and SDRPlay in my ubuntu 14 vm, but I already have the SoapySDR dependencies installed from source.

cjcliffe, 2015-11-13:
Based on my experience it looks like Ubuntu 14 is going to be the minimum for bundled releases for now.  May investigate this again later if there's demand.

closed 2015-11-13 by cjcliffe