cjcliffe/CubicSDR #160: Re-scan button on Device Selection screen (1.08+)

Toontje, 2015-10-14:
I didn't have my device connected so the device doesn't show up in the Device Selection screen.
Now i need to close CubicSDR, connect my device, start CubicSDR and then the device is selected.
A Re-scan button would eliminate the need to stop and re-launch CubicSDR.

cjcliffe, 2015-10-14:
I was close to having one already; it will stop any active stream since it has to free Soapy to re-scan and would probably crash some drivers otherwise; should be able to add this soon though.

Toontje, 2015-10-15:
No worries. At the moment a nice to have. Get the core right is more important and you're doing a great job at that.

Toontje, 2015-12-12:
I think this one can be closed, no?

cjcliffe, 2015-12-12:
Nope, still need to add a refresh button -- only remotes can technically refresh at the moment if you enter the IP twice :)

Toontje, 2015-12-13:
Does the SDR Devices menu option not do a new detection of connected
devices?

cjcliffe, 2015-12-13:
@Toontje nope not yet :)

cjcliffe, 2016-01-18:
There's now a 'Refresh' button on the Device Dialog in master, it'll stop active streams to prevent crashing but appears to be working well here plugging and unplugging devices or adding SoapyRemotes to the network.

closed 2016-01-18 by cjcliffe