cjcliffe/CubicSDR #161: Create/delete modulator by entering frequencies

Toontje, 2015-10-14:
It would be helpful if we could have a dialog where we can enter frequencies and modes with active/non-active checkboxes to activate/deactivate modulators.

cjcliffe, 2015-10-14:
Feel free to curate the checklist for this issue; I think your original suggestion is in there somewhere :)

AlmightyOatmeal, 2016-11-18:
I would absolutely love the ability to save bookmarks and be able to group them.

cjcliffe, 2016-11-21:
@AlmightyOatmeal already partly working in the bookmarks branch (minus actually saving); hopefully will have some early tests out in the next few weeks.

cjcliffe, 2016-12-30:
@Toontje OSX test build is up at https://github.com/cjcliffe/CubicSDR/releases/tag/0.2.1-alpha1 
Other platform builds will be ready soon

Toontje, 2016-12-30:
Perfect implementation of this function. For me you can close this issue.

cjcliffe, 2017-01-02:
@Toontje thanks 👍 closing this up; we can make new ones for specific issues.

closed 2017-01-02 by cjcliffe