cjcliffe/CubicSDR #162: Crash on opening XML preset

jacobseptember, 2015-10-15:
I'm on a Hackintosh, currently running OS X 10.11 El Capitan. This problem existed on 10.10 Yosemite, but didn't happen as often.

When opening an XML preset, nearly every time CubicSDR will crash. On Yosemite this seemed to happen more often when double-clicking on an item, instead of clicking on it and then clicking Open. On El Capitan it happens nearly every time; sometimes it works, but I haven't been able to notice a pattern.

An exemplary crash log: [link](http://www.cems.uvm.edu/~jseptemb/sdr/CubicSDR_015_presetCrash.txt)

cjcliffe, 2015-10-15:
@jacobseptember thanks I'll check this out; I haven't reviewed session loading since the previous release..

cjcliffe, 2015-11-27:
fixed several race conditions related to this and have eliminated loading crashes here, closing this one for now, thanks!

closed 2015-11-27 by cjcliffe