cjcliffe/CubicSDR #166: Add option to define path for SoapySDR

josh64x2, 2015-10-22:
On OS X, if you have installed SoapySDR and modules in a location other than `/usr/local/` (e.g. you used homebrew to install SoapySDR and you've installed homebrew in `/opt/`), CubicSDR fails to find the SDR devices.
It would be great to be able to define the installation path for SoapySDR and modules.

cjcliffe, 2015-10-22:
Hmm, SoapySDR includes it's own CMake find module; did you try checking 'Advanced' to see if the path shows there or did the CMake find module not get installed with brew SoapySDR?

josh64x2, 2015-10-22:
When I start CubicSDR 0.1.11 from the command line it prints out the following:
```
SDR enumerator starting.
SoapySDR init..
	API Version: v0.3.0-g9671ca26
	ABI Version: v0.3-0
	Install root: /usr/local
No modules found!
	Loading modules... done
	Available factories...null

Reporting enumeration complete.
SDR enumerator done.
```

Where is "Advanced" to check? I can't see that in CubicSDR...

Thanks!

cjcliffe, 2015-10-22:
Oh, sorry I see what you mean; SoapySDR actually lets you set "SOAPY_SDR_ROOT" environment variable.  I think you can also set it on startup like:

```
ccliffe$ SOAPY_SDR_ROOT="/opt/local" CubicSDR
```

josh64x2, 2015-10-23:
Excellent - Thanks!
```
SDR enumerator starting.
SoapySDR init..
	API Version: v0.3.0-g9671ca26
	ABI Version: v0.3-0
	Install root: /opt/homebrew
	Module found: /opt/homebrew/lib/SoapySDR/modules/librtlsdrSupport.so
	Loading modules... done
	Available factories...null, rtlsdr
```

cjcliffe, 2015-10-23:
Looks good, I'll add an option in CubicSDR to set that from the devices dialog and save it in the config; thanks for your help!

cjcliffe, 2015-11-12:
Kept this fairly simple; in latest commit to master SoapySDR module path can be specified with -m <module path> and will only attempt to load modules from that location.

Otherwise by default (with the next bundled release) installed modules at SOAPY_SDR_ROOT will be used first; followed by any bundled device modules.

closed 2015-11-12 by cjcliffe