cjcliffe/CubicSDR #167: Possibility to analyse bitstream

Toontje, 2015-10-23:
This is just a nice to have and a brainstorming opportunity.
Take a look at this video: https://www.dropbox.com/s/a655sw20o2klkvv/digital.mov?dl=0
Would it be possible to build "something" that allows us to analyse the bitstream? I am thinking at something like recording, maybe cleaning up/normalizing and playing back the bitstream and convert it into 0's and 1's. Maybe this is something that needs to be done outside CubicSDR so we need a way to get this data from CubicSDR into maybe Pothos (preferred) or gnuradio.
Just thinking.
Again, nice to have and low priority if any.

Toontje, 2015-10-28:
Something like this:
[image]

vsboost, 2015-10-28:
That would be real handy i must say.

cjcliffe, 2015-10-28:
This looks very similar to what I had in mind for the digital lab updates :)

Toontje, 2015-10-28:
That would make CubicSDR pass HDSDR in terms of functionality. Being a Mac user i am now permanently using CubicSDR and am never going back to Windows anymore unless it's for this kind of functionality. Let's make CubicSDR the new multi-platform HDSDR. 
BTW, what is digital lab?

cjcliffe, 2015-10-28:
@Toontje digital_lab branch incorporates the digital demodulation work already done by @Dantali0n just needs to be expanded to include more information and controls.   I plan to merge it in once I start refactoring and dividing some of the demodulation classes -- which is the next clean-up step I'd like to do.