cjcliffe/CubicSDR #168: CubicSDR with SoapySDR support don't compile on Ubuntu 14.04

frspin, 2015-10-26:
Latest CubicSDR from git repo, soapySDR support branch, don't compile on my Ubuntu 14.04 64 bit.
Compile problem is in WaterfallPanel.cpp

CubicSDR/src/panel/WaterfallPanel.cpp:87:9: error: ‘lines_buffered’ was not declared in this scope
         lines_buffered++;
         ^
CubicSDR/src/panel/WaterfallPanel.cpp: In member function ‘void WaterfallPanel::update()’:
CubicSDR/src/panel/WaterfallPanel.cpp:94:10: error: ‘bufferInitialized’ was not declared in this scope
     if (!bufferInitialized.load()) {
          ^
CubicSDR/src/panel/WaterfallPanel.cpp:98:10: error: ‘texInitialized’ was not declared in this scope
     if (!texInitialized.load()) {
          ^
CubicSDR/src/panel/WaterfallPanel.cpp:124:28: error: ‘lines_buffered’ was not declared in this scope
     for (int i = 0, iMax = lines_buffered.load(); i < iMax; i++) {
                            ^
CubicSDR/src/panel/WaterfallPanel.cpp:132:12: error: ‘lines_buffered’ was not declared in this scope
     while (lines_buffered.load()) {
            ^
CubicSDR/src/panel/WaterfallPanel.cpp: In member function ‘virtual void WaterfallPanel::drawPanelContents()’:
CubicSDR/src/panel/WaterfallPanel.cpp:154:10: error: ‘texInitialized’ was not declared in this scope
     if (!texInitialized.load()) {
          ^
make[2]: *** [CMakeFiles/CubicSDR.dir/src/panel/WaterfallPanel.cpp.o] Errore 1
make[1]: *** [CMakeFiles/CubicSDR.dir/all] Errore 2
make: *** [all] Errore 2

Previous version of CubicSDR, before mods to WaterfallPanel.ccp and WaterfallPanel.h (4 days ago) was compiled without problem.

Regards
Franco Spinelli
IW2DHW

cjcliffe, 2015-10-26:
Hey Franco,

Those files are definitely up to date on-line; It looks like there's something strange with your checked out repository -- might want to try cloning and building it again.

-CJ

frspin, 2015-10-27:
2015-10-26 22:30 GMT+01:00 Charles J. Cliffe <notifications@github.com>:

> Hey Franco,
>
> Those files are definitely up to date on-line; It looks like there's
> something strange with your checked out repository -- might want to try
> cloning and building it again.
>
>
Already done.
But also from web view of github repo I get for WaterfallPanel.cpp and
WaterfallPanel.h an update to 5 days ago.
And this is my downloaded version, not compiling in Ubuntu 14.04.

Same downloading with "Download ZIP" button.  Tested on 2 different Ubuntu
14.04/64 bit PC
A Github problem?

Regards

Franco Spinelli
IW2DHW

jazzkutya, 2015-10-27:
i have the same problem
tried checking out soapysdr-support branch
also tried merging soapysdr-support branch to master
both build died with same problem

jazzkutya, 2015-10-27:
In file included from /home/vrm/sdrplay/cubicSDR/CubicSDR/src/panel/WaterfallPanel.cpp:1:0:
/home/vrm/sdrplay/cubicSDR/CubicSDR/src/panel/WaterfallPanel.h:27:5: error: ‘atomic_int’ in namespace ‘std’ does not name a type
     std::atomic_int lines_buffered;

this is the main culprit

jazzkutya, 2015-10-27:
Charles, you forgot to #include <atomic> in WaterfallPanel.h :)
I have put the include there and now it builds fine.

frspin, 2015-10-27:
> Charles, you forgot to #include in WaterfallPanel.h :)
> I have put the include there and now it builds fine.

Please attach correct #include in WaterfallPanel.h

Regards

Franco Spinelli

jazzkutya, 2015-10-27:
I can't attach anything but PNG, GIF, JPG. No permission.

jazzkutya, 2015-10-27:
test &gt; &lt;

jazzkutya, 2015-10-27:
previous comment edited to show the solution

jazzkutya, 2015-10-27:
here is the correct top of that file:
#pragma once

#include "GLPanel.h"
#include &lt;atomic&gt;

class WaterfallPanel : public GLPanel {

frspin, 2015-10-27:
2015-10-27 17:09 GMT+01:00 jazzkutya <notifications@github.com>:

> previous comment edited to show the solution
>
>
OK
Now compile correct.
This evening I will test with SDRPlay

Regards
Franco Spinelli
IW2DHW

cjcliffe, 2015-10-27:
@jazzkutya @frspin thanks; fix committed to soapysdr-support branch :+1:

frspin, 2015-10-27:
Il 27/10/2015 17:19, Charles J. Cliffe ha scritto:
> @jazzkutya <https://github.com/jazzkutya> @frspin
> <https://github.com/frspin> thanks; fix committed to soapysdr-support
> branch :+1:
> 
> —

Now the problem is on startup of CubicSDR.

With a RTLSDR and with a SDRPlay CubicSDR start and, after selecting
correct device, core dump.

Here is last part of messages on terminal window:

SoapySDR init..
	API Version: v0.4.0-ge62dba9d
	ABI Version: v0.4-2
	Install root: /usr/local
	Module found: /usr/local/lib/SoapySDR/modules/libairspySupport.so
	Module found: /usr/local/lib/SoapySDR/modules/libremoteSupport.so
	Module found: /usr/local/lib/SoapySDR/modules/librfspaceSupport.so
	Module found: /usr/local/lib/SoapySDR/modules/librtlsdrSupport.so
	Module found: /usr/local/lib/SoapySDR/modules/libsdrPlaySupport.so
	Loading modules... done
	Available factories...airspy, null, remote, rfspace, rtlsdr, sdrplay
mir_sdr_Init: starting hardware initialization
mir_sdr_Init: gR=40dB fs=2.048MHz rf=222.064MHz bw=1.536MHz if=0.000MHz

mir_sdr_usb_USB DLL: Revision 0.1.1

Warning: libusb_claim_interface() -6

mir_sdr_2500_Init: fnaddr = 2 detected, trying to change...

mir_sdr_2500_Init: fnaddr = 6

mir_sdr_2500_Init: adjusting squelch trim 0x1, rx gating enable 1,
tx_trim 0, reg2 = 0x4801
initHw: Register7 = 0x000005
initHw: Tuner Register0 = 0x04f440
mir_sdr_SetFs: Sample Freq requested 2047999.978065
mir_sdr_SetFs: Fs->FsNomHz+dFsHz=2048000.0+0.0Hz=2048000.0Hz FsToggle->1
mir_sdr_SetRf: f->222064000.000Hz (int=25 frac=20 afc=0)
fSynth:3553024000.000
mir_sdr_SetRf:
Rf->RfNomHz+dRfHz+LifHz+Lif1Hz=222064000.0+0.0+0.0Hz+0.0Hz=222064000.0Hz
RfToggle->1
mir_sdr_SetGr: GR->40[16,24,0,0] gRset->0x210 DCCALmode=4 DCCALspd=1
GrToggle->1
setToggleStates: initialising sampNum=0x0001f802, gainSetting=0x210,
FsToggle=0, RfToggle=1, GrToggle=1
setToggleStates: initialising Fs=2047999.978, Rf=222064000.000, Gr=40
mir_sdr_2500_Readback001Trim: readback = 0xe30160f
mir_sdr_2500_Bypass001Trim: reg13=0x1ed reg14=0x6c016e
  driver = sdrplay
  label = SDRPlay RSP
Make device 0
mir_sdr_Init: starting hardware initialization
mir_sdr_Init: gR=40dB fs=2.048MHz rf=222.064MHz bw=1.536MHz if=0.000MHz

mir_sdr_usb_USB DLL: Revision 0.1.1

Warning: libusb_claim_interface() -6

mir_sdr_2500_Init: fnaddr = 2 detected, trying to change...

mir_sdr_2500_Init: fnaddr = 6

mir_sdr_2500_Init: adjusting squelch trim 0x1, rx gating enable 1,
tx_trim 0, reg2 = 0x4801
initHw: Register7 = 0x000005
initHw: Tuner Register0 = 0x04f440
mir_sdr_SetFs: Sample Freq requested 2047999.978065
mir_sdr_SetFs: Fs->FsNomHz+dFsHz=2048000.0+0.0Hz=2048000.0Hz FsToggle->1
mir_sdr_SetRf: f->222064000.000Hz (int=25 frac=20 afc=0)
fSynth:3553024000.000
mir_sdr_SetRf:
Rf->RfNomHz+dRfHz+LifHz+Lif1Hz=222064000.0+0.0+0.0Hz+0.0Hz=222064000.0Hz
RfToggle->1
mir_sdr_SetGr: GR->40[16,24,0,0] gRset->0x210 DCCALmode=4 DCCALspd=1
GrToggle->1
setToggleStates: initialising sampNum=0x0001f802, gainSetting=0x210,
FsToggle=0, RfToggle=0, GrToggle=0
setToggleStates: initialising Fs=2047999.978, Rf=222064000.000, Gr=40
mir_sdr_2500_Readback001Trim: readback = 0xe30160f
mir_sdr_2500_Bypass001Trim: reg13=0x1ed reg14=0x6c016e
  mir_sdr_version=1.700000

Reporting enumeration complete.
SDR enumerator done.
Set sample rate: 2000000
Set PPM: 0
Set direct sampling mode: 0
Set offset: 0
SDR thread starting.
device init()
mir_sdr_Init: starting hardware initialization
mir_sdr_Init: gR=40dB fs=2.048MHz rf=222.064MHz bw=1.536MHz if=0.000MHz

mir_sdr_usb_USB DLL: Revision 0.1.1

Errore di segmentazione (core dump creato)

Removing ~/.CubicSDR directory don't help.
Previous version worked without problems.

Regards

Franco Spinelli

cjcliffe, 2015-10-27:
I've committed some additional fixes that fix some startup and runtime crashes in windows, going to try it here on ubuntu linux shortly.  In the meantime make sure SoapySDR and SoapySDRPlay are up to date as there's some recent fixes there as well.

frspin, 2015-10-28:
Il 27/10/2015 22:23, Charles J. Cliffe ha scritto:
> I've committed some additional fixes that fix some startup and runtime
> crashes in windows, going to try it here on ubuntu linux shortly. In the
> meantime make sure SoapySDR and SoapySDRPlay are up to date as there's
> some recent fixes there as well.
> 

There is something strange.

Without any device connected CubicSDR start, I dismiss initial pop-up
window and after this I can use it, obviously without any signal. Menu',
GUI, setup all are working.
Only uncheck of Menu'->Setting->Automatic gain produce a core dump.

If I connect a device, after initial pop-up of SDR Device choice, there
is immediately a core dump.

I have reloaded from GitHub all CubicSDR and Soapy related files.
I have also reloades liquid-dsp from GitHub.

But the problem is still here.

If you want me to do some testing, I can do in the evening.

Regards
Franco Spinelli
IW2DHW

cjcliffe, 2015-10-28:
@frspin the gain problem makes sense; I haven't accounted for the fact that you might switch that while the device isn't connected..  I'll have to patch that up.

To help find the other crash If you can create "Debug" targets (-DCMAKE_BUILD_TYPE=Debug or set "Build Type" to "Debug" in the CMake GUI for CubicSDR) and the same for the SoapySDR libs and then build everything (note: make sure to build and install SoapySDR first!) and run it with 'gdb':

```
gdb ./CubicSDR
  == bunch of loading messages and stuff ==
(gdb) run
  == CubicSDR runs here.. then segfaults ==
(gdb) backtrace
  == {This is the result I'll need to see..} ==
```

Thanks!

SDRplay, 2015-10-28:
@cjcliffe I got the same behaviour as @frspin but when I compile with debug I get a slightly different behaviour. It now actually lets you select the device, but when you select it, I get...

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff34d6fe0 in g_str_hash () from /lib/x86_64-linux-gnu/libglib-2.0.so.0

I also got segmentation fault when running SoapySDRUtil but struggling to get it to run in gdb - how to do you run with command line options in gdb? It thinks I'm trying to pass the options to the debugger not the software under test. Quotes don't seem to work either.

SDRplay, 2015-10-28:
If I run the gdb ./CubicSDR as root then it actually runs and I get a display and audio. Then after about a minute, I get...

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffb65fc700 (LWP 30484)]
__GI___pthread_mutex_lock (mutex=0x7fffac012ee0)
    at ../nptl/pthread_mutex_lock.c:66
66	../nptl/pthread_mutex_lock.c: No such file or directory.

SDRplay, 2015-10-28:
ok got SoapySDRUtil to run in gdb...

Starting program: /usr/local/bin/SoapySDRUtil --probe=sdrplay
Traceback (most recent call last):
  File "/usr/share/gdb/auto-load/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.19-gdb.py", line 63, in <module>
    from libstdcxx.v6.printers import register_libstdcxx_printers
ImportError: No module named 'libstdcxx'
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Probe device sdrplay
mir_sdr_Init: starting hardware initialization
mir_sdr_Init: gR=40dB fs=2.048MHz rf=222.064MHz bw=1.536MHz if=0.000MHz

mir_sdr_usb_USB DLL: Revision 0.1.1

[New Thread 0x7ffff3d23700 (LWP 30546)]
Warning: libusb_claim_interface() -6

[New Thread 0x7ffff3522700 (LWP 30547)]
mir_sdr_2500_Init: revisionId = 0x0200, doing FW update
fwDownload: FW image size = 6008
[Thread 0x7ffff3522700 (LWP 30547) exited]
[Thread 0x7ffff3d23700 (LWP 30546) exited]
mir_sdr_usb_USB DLL: Revision 0.1.1

[New Thread 0x7ffff3d23700 (LWP 30554)]
Warning: libusb_claim_interface() -6

[New Thread 0x7ffff3522700 (LWP 30555)]
mir_sdr_2500_Init: fnaddr = 2 detected, trying to change...

mir_sdr_2500_Init: fnaddr = 18

mir_sdr_2500_Init: adjusting squelch trim 0x1, rx gating enable 1, tx_trim 0, reg2 = 0x4801
initHw: Register7 = 0x000005
initHw: Tuner Register0 = 0x04f440
mir_sdr_SetFs: Sample Freq requested 2047999.978065
mir_sdr_SetFs: Fs->FsNomHz+dFsHz=2048000.0+0.0Hz=2048000.0Hz FsToggle->1
mir_sdr_SetRf: f->222064000.000Hz (int=25 frac=20 afc=0) fSynth:3553024000.000
mir_sdr_SetRf: Rf->RfNomHz+dRfHz+LifHz+Lif1Hz=222064000.0+0.0+0.0Hz+0.0Hz=222064000.0Hz RfToggle->1
mir_sdr_SetGr: GR->40[16,24,0,0] gRset->0x210 DCCALmode=4 DCCALspd=1 GrToggle->1
setToggleStates: initialising sampNum=0x0001f802, gainSetting=0x210, FsToggle=0, RfToggle=1, GrToggle=1
setToggleStates: initialising Fs=2047999.978, Rf=222064000.000, Gr=40
mir_sdr_2500_Readback001Trim: readback = 0xe381613
mir_sdr_2500_Bypass001Trim: reg13=0x26d reg14=0x6e016e
[Thread 0x7ffff3522700 (LWP 30555) exited]
[Thread 0x7ffff3d23700 (LWP 30554) exited]

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRPlay
  hardware=SDRPlay
  mir_sdr_version=1.700000

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: NO
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=2048]
  Antennas: RX
  Corrections: DC removal, DC offset
  Full gain range: [0, 204] dB
    TUNER gain range: [0, 102] dB
    LNAT gain range: [0, 102] dB
  Full freq range: [0.1, 2000] MHz
    RF freq range: [0.1, 2000] MHz
  Sample rates: [1.536, 12] MHz
  Filter bandwidths: [0.2, 8] MHz

[Inferior 1 (process 30542) exited normally]
(gdb) quit

error at the top, but other than that ok - that's not what happens when compiled without debug, it segmentation faults unless run as root

cjcliffe, 2015-10-28:
@SDRplay I see a note on facebook that says it will segfault if it's not readable/executable by the current user -- Can you check the SoapySDR and SoapySDRPlay file permissions?

SDRplay, 2015-10-28:
-rw-r--r-- 1 root root 662833 Oct 28 22:00 libsdrPlaySupport.so
-rw-r--r-- 1 root root 1582631 Oct 28 22:00 /usr/local/lib/libSoapySDR.so.0.4.0
-rwxr-xr-x 1 root root 343851 Oct 28 22:00 /usr/local/bin/SoapySDRUtil

SDRplay, 2015-10-28:
I've just rebuild SoapySDR, SoapySDRPlay and CubicSDR all with -DCMAKE_BUILD_TYPE=Release and now SoapySDRUtil runs ok as a user. CubicSDR will only run as root, otherwise I get a segmentation fault, but as root it runs fine, finds the device and displays signal fine.

cjcliffe, 2015-10-28:
@SDRplay that sounds strange.. working as root still indicates some sort of USB device or file permissions still I think..  Does SDRPlay need some udev rules or something?

SDRplay, 2015-10-28:
rules file: SUBSYSTEM=="usb",ENV{DEVTYPE}=="usb_device",ATTRS{idVendor}=="1df7",ATTRS{idProd
uct}=="2500",MODE:="0666"

frspin, 2015-10-29:
2015-10-28 23:58 GMT+01:00 SDRplay <notifications@github.com>:

> rules file:
> SUBSYSTEM=="usb",ENV{DEVTYPE}=="usb_device",ATTRS{idVendor}=="1df7",ATTRS{idProd
> uct}=="2500",MODE:="0666"
>
>
After all this tests I have done some other test using an RTLSDR dongle

First I have compiled all Soapy libraries and CubicSDR with
-DCMAKE_BUILD_TYPE=Release

After this:

rtl_test run OK as user, so no problem with UDEV rules
SoapySDRUtil --probe run OK as user and find my dongle
CubicSDR crash if runned as user
CubicSDR run OK an root, also after few minutes

So the problem seem is in CubicSDR, after selecting the device and before
starting the selected device.

Version of previous week of CubicSDR and SoapySDR libraries was working
without any problem with same hardware.

Regards

Franco Spinelli
IW2DHW

frspin, 2015-10-29:
2015-10-28 22:58 GMT+01:00 Charles J. Cliffe <notifications@github.com>:

> gdb ./CubicSDR == bunch of loading messages and stuff == (gdb) run ==
> CubicSDR runs here.. then segfaults == (gdb) backtrace == {This is the
> result I'll need to see..} ==
>

As required, this is backtrace output of CubicSDR running with a RTLSDR
dongle:

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff34d6fe0 in g_str_hash () from
/lib/x86_64-linux-gnu/libglib-2.0.so.0
(gdb) backtrace
#0  0x00007ffff34d6fe0 in g_str_hash ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#1  0x00007ffff34d6040 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007fffea061162 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libunity-gtk2-parser.so.0
#3  0x00007fffea05a8e2 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libunity-gtk2-parser.so.0
#4  0x00007fffea05c6ad in ?? ()
   from /usr/lib/x86_64-linux-gnu/libunity-gtk2-parser.so.0
#5  0x00007ffff44713b8 in g_closure_invoke ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#6  0x00007ffff4482d3d in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#7  0x00007ffff448aa29 in g_signal_emit_valist ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#8  0x00007ffff448ace2 in g_signal_emit ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#9  0x00007ffff4475725 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#10 0x00007ffff4477ceb in g_object_notify ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#11 0x00007ffff4ba7e05 in gtk_widget_show ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
---Type <return> to continue, or q <return> to quit---
#12 0x00007ffff701c865 in wxMenu::GtkAppend(wxMenuItem*, int) ()
   from /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#13 0x00007ffff701cd66 in wxMenu::DoAppend(wxMenuItem*) ()
   from /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#14 0x0000000000489d82 in wxMenuBase::Append (this=0x2e02450, itemid=2150,
    text=..., help=..., kind=wxITEM_RADIO) at
/usr/include/wx-3.0/wx/menu.h:67
#15 0x0000000000489e9e in wxMenuBase::AppendRadioItem (this=0x2e02450,
    itemid=2150, text=..., help=...) at /usr/include/wx-3.0/wx/menu.h:86
#16 0x000000000048336f in AppFrame::initDeviceParams (this=0x8bb660,
    devInfo=0x7fffb0009c00)
    at
/home/spin/Dati/download/linux/ham/CubicSDR/CubicSDR/src/AppFrame.cpp:451
#17 0x0000000000476368 in CubicSDR::setDevice (this=0x8453c0,
    dev=0x7fffb0009c00)
    at
/home/spin/Dati/download/linux/ham/CubicSDR/CubicSDR/src/CubicSDR.cpp:450
#18 0x0000000000524a4a in SDRDevicesDialog::OnUseSelected (this=0x36e0e00,
    event=...)
    at
/home/spin/Dati/download/linux/ham/CubicSDR/CubicSDR/src/forms/SDRDevices/SDRDevices.cpp:55
#19 0x00007ffff692311e in wxAppConsoleBase::CallEventHandler(wxEvtHandler*,
wxEventFunctor&, wxEvent&) const ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
---Type <return> to continue, or q <return> to quit---
#20 0x00007ffff6a96282 in
wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&,
wxEvtHandler*, wxEvent&) ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#21 0x00007ffff6a965e6 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#22 0x00007ffff6a9666e in wxEvtHandler::TryHereOnly(wxEvent&) ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#23 0x00007ffff6a96703 in wxEvtHandler::ProcessEventLocally(wxEvent&) ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#24 0x00007ffff6a96765 in wxEvtHandler::ProcessEvent(wxEvent&) ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#25 0x00007ffff6a964f7 in wxEvtHandler::SafelyProcessEvent(wxEvent&) ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#26 0x00007ffff6fcc5ad in ?? ()
   from /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#27 0x00007ffff4a92815 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#28 0x00007ffff44713b8 in g_closure_invoke ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#29 0x00007ffff4482d3d in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#30 0x00007ffff448a6f9 in g_signal_emit_valist ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
---Type <return> to continue, or q <return> to quit---
#31 0x00007ffff448ace2 in g_signal_emit ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#32 0x00007ffff4ba26b4 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#33 0x00007ffff4a90fc4 in gtk_propagate_event ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#34 0x00007ffff4a9137b in gtk_main_do_event ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#35 0x00007ffff470b43c in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#36 0x00007ffff34e6e04 in g_main_context_dispatch ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#37 0x00007ffff34e7048 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#38 0x00007ffff34e730a in g_main_loop_run ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#39 0x00007ffff4a90447 in gtk_main ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#40 0x00007ffff6faf145 in wxGUIEventLoop::DoRun() ()
   from /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#41 0x00007ffff6965440 in wxEventLoopBase::Run() ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#42 0x00007ffff69251fd in wxAppConsoleBase::MainLoop() ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
---Type <return> to continue, or q <return> to quit---
#43 0x00007ffff69b104d in wxEntry(int&, wchar_t**) ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#44 0x0000000000473eae in main (argc=1, argv=0x7fffffffdd68)
    at
/home/spin/Dati/download/linux/ham/CubicSDR/CubicSDR/src/CubicSDR.cpp:22

Hope this help

Franco Spinelli
IW2DHW

cjcliffe, 2015-10-29:
@frspin thanks; not quite where I expected it to be crashing -- looks like it crashed somewhere in wxWidgets while attempting to update/add the device input rate menu items..  I should be able to think up a workaround for that; will update issue when patched.

frspin, 2015-10-30:
2015-10-29 17:45 GMT+01:00 Charles J. Cliffe <notifications@github.com>:

> @frspin <https://github.com/frspin> thanks; not quite where I expected it
> to be crashing -- looks like it crashed somewhere in wxWidgets while
> attempting to update/add the device input rate menu items.. I should be
> able to think up a workaround for that; will update issue when patched.
>
>
If this can help, this problem appeared for first time friday 23 october
evening.

Regards
Franco Spinelli
IW2DHW

frspin, 2015-10-31:
Il 29/10/2015 17:45, Charles J. Cliffe ha scritto:
> @frspin <https://github.com/frspin> thanks; not quite where I expected
> it to be crashing -- looks like it crashed somewhere in wxWidgets while
> attempting to update/add the device input rate menu items.. I should be
> able to think up a workaround for that; will update issue when patched.
> 
> —

Meanwhile CubicSDR works great with SDRPlay for receiving Voa Radiogram,
obviously running with "sudo" as root.

Regards

Franco Spinelli
IW2DHW

frspin, 2015-11-02:
Il 29/10/2015 17:45, Charles J. Cliffe ha scritto:
> @frspin <https://github.com/frspin> thanks; not quite where I expected
> it to be crashing -- looks like it crashed somewhere in wxWidgets while
> attempting to update/add the device input rate menu items.. I should be
> able to think up a workaround for that; will update issue when patched.
> 

I have done some test but I don't know C++ so my test are "elementary test"

Segmentation fault happen at line 460 of AppFrame.cpp, when there is
first allocation of

sampleRateMenuItems[wxID_BANDWIDTH_BASE+ofs] =
sampleRateMenu->AppendRadioItem(wxID_BANDWIDTH_BASE+ofs,
frequencyToStr(*i));

Removing this statement and also statements at line 462

sampleRateMenuItems[wxID_BANDWIDTH_BASE+ofs]->Check(true);

and at line 468

sampleRateMenuItems[wxID_BANDWIDTH_MANUAL] =
sampleRateMenu->AppendRadioItem(wxID_BANDWIDTH_MANUAL, "Manual Entry");

I get an empty Input Bandwidth menu but no Segmentation fault.

Can be an uninitialized (or badly initialized) area? And why running as
root this error don't appear?

In older version, running as user, a change in this menu  caused a
segmentation error. No problem at all running as root.

Regards

Franco Spinelli
IW2DHW

cjcliffe, 2015-11-02:
@frspin thanks that should help me narrow it down but I haven't been able to reproduce it here so far; can you confirm the version of wxWidgets you're using is at least 3.0.2?

frspin, 2015-11-02:
Il 02/11/2015 18:50, Charles J. Cliffe ha scritto:
> @frspin <https://github.com/frspin> thanks that should help me narrow it
> down but I haven't been able to reproduce it here so far; can you
> confirm the version of wxWidgets you're using is at least 3.0.2?
> 

No. Ubuntu 14.04 have Wx 2.8 and Wx 3.0.0
CubicSDR is using Wx 3.0.0

And any operation on sampleRateMenu is impossible and go to a
segmentation error.

Also

    for (std::map<int, wxMenuItem *>::iterator i =
sampleRateMenuItems.begin(); i != sampleRateMenuItems.end(); i++) {
        sampleRateMenu->Remove(i->first);
    }

is not executed. Both begin() and end() are 0, so no operation on
sampleRateMenu

sampleRates.begin() is 256000 for my RTLSDR and sampleRates.end() is 96.
Is this correct?

Can be a WX version problem?

If needed, I can put up a PC with Ubuntu 15.10 version and do some test
on it.

Regards

Franco Spinelli
IW2DHW

frspin, 2015-11-02:
Il 02/11/2015 18:50, Charles J. Cliffe ha scritto:
> @frspin <https://github.com/frspin> thanks that should help me narrow it
> down but I haven't been able to reproduce it here so far; can you
> confirm the version of wxWidgets you're using is at least 3.0.2?
> 

I have installed Ubuntu 15.10/64 bit on an old machine for test only.

Kernel is 4.2.0 and WxWidget is 3.0.2.

I have used all libraries from repo, downloaded liquid-dsp, SoapySDR and
SoapyRTLSDR from GitHub and compiled. After this I have downloaded and
compiled CubicSDR and there is no segmentation fault running it as user.

So the problem is in WxWidget of Ubuntu 14.04? Which changes was made to
previous running version?

Regards

Franco Spinelli
IW2DHW

cjcliffe, 2015-11-02:
@frspin I'm guessing it's a bug fixed between 3.0.0 and 3.0.2; the major difference in CubicSDR is that the bandwidth menu is now dynamic for the device; the values were fixed before and set before the window was loaded; now they're set after the window has loaded and the device has initialized.

I'm not sure it's worth a workaround unless there's a multitude of systems running 3.0.0; I can just make sure that the requirements are 3.0.2 so that it's at least clear during cmake setup.

cjcliffe, 2015-11-03:
@frspin we've successfully traced and fixed this issue now for Ubuntu 14; apparently something doesn't like a drop-down menu being updated dynamically -- but it's ok with replacing the entire menu with a new one.  

So not a wxWidgets 3.0.2 issue but updating OS apparently does fix it so something in the WM is likely upset with changing contents of active menus.

frspin, 2015-11-03:
Il 03/11/2015 04:16, Charles J. Cliffe ha scritto:
> @frspin <https://github.com/frspin> we've successfully traced and fixed
> this issue now for Ubuntu 14; apparently something doesn't like a
> drop-down menu being updated dynamically -- but it's ok with [...]

Updated, compiled and tested

This solve the problem. Thank you

Now, for USB/LSB reception, I will do more test and try different bandwidth.

But this is another question and, if necessary, I will open another issue.

Regards
Franco Spinelli
IW2DHW

cjcliffe, 2015-11-03:
@frspin glad to hear it's working; I'll close this one for now; and feel free to open additional issues.

closed 2015-11-03 by cjcliffe