cjcliffe/CubicSDR #169: Manual gain control for RTLSDR devices

dc1rdb, 2015-10-28:
Not sure if this is a CubicSDR issue or a SoapyRTLSDR issue.

Just tried the new manual gain mode with three different RTLSDR dongles:
1) Generic R820 tuner: manual gain setting works as expected

2) Elonics E4000 tuner: six IF gain sliders instead of one. Neither the IF gain sliders nor the tuner gain sliders has any effect.
[image]

3) Fitipower FC0013 tuner: Zero position is in the lower third of the scale, gain increases with slider movement down (negative gain values). Even with slider all the way down (indicating -9), gain is lower than in AGC mode. Moving slider above zero position has no effect.
[image]

cjcliffe, 2015-10-28:
@dc1rdb E4000 problem makes sense; I only have 820T / T2 here so I just followed the "E4000 has 6 IF stages" in the headers..  FC0013 seems weird; I might just be reading the gain limits wrong -- what does ```rtl_test``` vs. ```SoapySDRUtil --probe``` program report for available gains on each device?

dc1rdb, 2015-10-28:
Fitipower FC0013:
```
mb-rd:~ rduering$ rtl_test
Found 1 device(s):
  0:  NOXON, DAB Stick, SN: 0

Using device 0: Terratec NOXON DAB/DAB+ USB dongle (rev 1)
Found Fitipower FC0013 tuner
Supported gain values (23): -9.9 -7.3 -6.5 -6.3 -6.0 -5.8 -5.4 5.8 6.1 6.3 6.5 6.7 6.8 7.0 7.1 17.9 18.1 18.2 18.4 18.6 18.8 19.1 19.7 
Sampling at 2048000 S/s.
```

```
mb-rd:~ rduering$ SoapySDRUtil --probe
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Probe device 
Found Fitipower FC0013 tuner
mir_sdr_Init: starting hardware initialization
mir_sdr_Init: gR=40dB fs=2.048MHz rf=222.064MHz bw=1.536MHz if=0.000MHz

mir_sdr_usb_USB DLL: Revision 1.8.0

mir_sdr_usb_Init: Timeout expired/failed to establish connection with the device

mir_sdr_2500_Init: mir_sdr_usb_Init() failed
mir_sdr_Init: mir_sdr_2500_Init() Error 1
Found Fitipower FC0013 tuner

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=RTLSDR
  hardware=RTLSDR
  origin=https://github.com/pothosware/SoapyRTLSDR
  rtl=0

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * Direct Sampling - RTL-SDR Direct Sampling Mode, 0 = Off, 1 = I ADC, 2 = Q ADC
       [key=direct_samp, default=0, type=int, range=[0, 2]]
     * Offset Tune - RTL-SDR Offset Tuning Mode
       [key=offset_tune, default=false, type=bool, range=[0, 2]]
     * I/Q Swap - RTL-SDR I/Q Swap Mode
       [key=iq_swap, default=false, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS8, CS16, CF32
  Native format: CS8 [full-scale=128]
  Stream args:
     * Buffer Size - Number of bytes per buffer, multiples of 512 only.
       [key=bufflen, units=bytes, default=16384, type=int]
     * Buffer Count - Number of buffers per read.
       [key=buffers, units=buffers, default=15, type=int]
  Antennas: RX
  Full gain range: [0, 29.6] dB
    TUNER gain range: [-9.9, 19.7] dB
  Full freq range: [26.999, 1764] MHz
    RF freq range: [27, 1764] MHz
    CORR freq range: [-0.001, 0.001] MHz
  Sample rates: [0.25, 3.2] MHz
```

Elonics E4000:
```
mb-rd:~ rduering$ rtl_test
Found 1 device(s):
  0:  Realtek, RTL2832U, SN: 00000001

Using device 0: Terratec NOXON DAB/DAB+ USB dongle (rev 2)
Found Elonics E4000 tuner
Supported gain values (14): -1.0 1.5 4.0 6.5 9.0 11.5 14.0 16.5 19.0 21.5 24.0 29.0 34.0 42.0 
Sampling at 2048000 S/s.
```

```
mb-rd:~ rduering$ SoapySDRUtil --probe
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Probe device 
Found Elonics E4000 tuner
mir_sdr_Init: starting hardware initialization
mir_sdr_Init: gR=40dB fs=2.048MHz rf=222.064MHz bw=1.536MHz if=0.000MHz

mir_sdr_usb_USB DLL: Revision 1.8.0

mir_sdr_usb_Init: Timeout expired/failed to establish connection with the device

mir_sdr_2500_Init: mir_sdr_usb_Init() failed
mir_sdr_Init: mir_sdr_2500_Init() Error 1
Found Elonics E4000 tuner

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=RTLSDR
  hardware=RTLSDR
  origin=https://github.com/pothosware/SoapyRTLSDR
  rtl=0

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * Direct Sampling - RTL-SDR Direct Sampling Mode, 0 = Off, 1 = I ADC, 2 = Q ADC
       [key=direct_samp, default=0, type=int, range=[0, 2]]
     * Offset Tune - RTL-SDR Offset Tuning Mode
       [key=offset_tune, default=false, type=bool, range=[0, 2]]
     * I/Q Swap - RTL-SDR I/Q Swap Mode
       [key=iq_swap, default=false, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS8, CS16, CF32
  Native format: CS8 [full-scale=128]
  Stream args:
     * Buffer Size - Number of bytes per buffer, multiples of 512 only.
       [key=bufflen, units=bytes, default=16384, type=int]
     * Buffer Count - Number of buffers per read.
       [key=buffers, units=buffers, default=15, type=int]
  Antennas: RX
  Full gain range: [0, 301] dB
    IF1 gain range: [-1, 42] dB
    IF2 gain range: [-1, 42] dB
    IF3 gain range: [-1, 42] dB
    IF4 gain range: [-1, 42] dB
    IF5 gain range: [-1, 42] dB
    IF6 gain range: [-1, 42] dB
    TUNER gain range: [-1, 42] dB
  Full freq range: [26.999, 1764] MHz
    RF freq range: [27, 1764] MHz
    CORR freq range: [-0.001, 0.001] MHz
  Sample rates: [0.25, 3.2] MHz
```

jazzkutya, 2015-10-28:
Charles, all the IF stages of e4k accept only gains from a limited set. And the usable values depend also on which stage are trying to set. From the rtlsdr e4k tuner code:
static const int8_t if_stage1_gain[] = {
        -3, 6
};

static const int8_t if_stage23_gain[] = {
        0, 3, 6, 9
};

static const int8_t if_stage4_gain[] = {
        0, 1, 2, 2
};

static const int8_t if_stage56_gain[] = {
        3, 6, 9, 12, 15, 15, 15, 15
};

if_stage23_gain applies to stage 2 and 3, applies to stage 5 and 6

the numbers are dB

cjcliffe, 2015-10-28:
Ok, I think I'll just have to set the minimum and maximum gains for those values and have the SoapyRTLSDR handle snapping to the nearest value..  

Also are the IF gains even worth having there?  or is that just going to make it confusing to use an E4k tuner?

jazzkutya, 2015-10-28:
I think SoapyRTLSDR will not snap to nearest value (judging by a brief look at it's code). Also librtlsdr just does not provide any way to query allowed values for the IF stages.

That's a good question, it definitely seems like a big hassle with implementation. I don't know how useful it is as i don't have rtlsdr with E4k tuner.

dc1rdb, 2015-10-29:
Similar gain setting problems were encountered during development of RF Analyzer: https://github.com/demantz/RFAnalyzer/issues/4
https://github.com/demantz/RFAnalyzer/commit/7b86fabd5cff86248279144a30fb8ccdaceaa66e
Maybe this helps...

cjcliffe, 2015-12-08:
Added issue to fix this up over at https://github.com/pothosware/SoapyRTLSDR/issues/18

cjcliffe, 2016-01-20:
@dc1rdb latest SoapyRTLSDR now has working E4000 gain control; let me know if that works for you when you have a chance.

dc1rdb, 2016-01-20:
Hmm, is it supposed to work with CubicSDR v0.1.20? I'm getting strange results here. IF1-4 gain sliders have very little effect, IF5-6 and tuner gain have no effect at all.
Screen recording at http://wikisend.com/download/523038/CubicSDR_E4000.m4v

dc1rdb, 2016-01-20:
With the Fitipower FC00013 tuner, the gain slider has very little effect, too: http://wikisend.com/download/256692/CubicSDR_FC00013.m4v

R820 manual tuner gain works as expected: http://wikisend.com/download/517368/CubicSDR_R820.m4v

cjcliffe, 2016-01-20:
@dc1rdb you'll need to build and install latest SoapyRTLSDR at least; there's some minor fixes for gain control in 0.1.21 (master branch) as well but shouldn't be required.

Once you've got the new version installed the IF gain sliders should be very short range except for the TUNER; there's only a limited set of values that the IFs can be but they're definitely having effects here on E4000.

Will do some more experiments here on FC00013 too, seemed to be working but just not particularly well; possibly just have the range wrong.

dc1rdb, 2016-01-21:
Hi Charles, just to reconfirm: all of the above tests were done with a fresh build of SoapyRTLSDR (latest commit a44abc3) and CubicSDR v0.1.20
Same result after replacing
```
/Applications/CubicSDR.app/Contents/MacOS/modules/librtlsdrSupport.so
```
with the newly built version.
Am I missing something?

cjcliffe, 2016-01-21:
You're probably not missing anything -- I'll re-investigate as I'm having some intermittent E4000 issues here myself where it randomly loses the PLL lock

closed 2016-08-17 by cjcliffe