cjcliffe/CubicSDR #170: Panoramic Mode

cjcliffe, 2015-10-30:
Panoramic mode will basically be the opposite of zooming-in.  If pano mode is enabled zooming-out will disable demodulation and begin panoramic scanning mode.  While zoomed out pressing space will bring up a dialog to set the current pano range instead of the center frequency.

- [ ] Allow enabling panoramic mode via menu
- [ ] zooming-out on the waterfall will engage panoramic mode
- [ ] zooming-in to a usable bandwidth will re-enable demodulation 
- [ ] pressing space while panoramic mode is engaged will display the range input dialog instead of center frequency

Dantali0n, 2016-03-20:
Yes this can suffice for a spectrum analyzer like feature however 2 important attributes are missing. 

Since noise sources and even tracking-generators do not create a flat frequency response. _A calibration/set relative feature_ is required which will take an initial output measurements at a given frequency and then from that point associate that output with 0dm.
I made this gif to demonstrate the behavior described above: [Gif link](https://imgflip.com/gif/114vur)
(Most spectrum analyzer also take multiple readings for every frequency interval and then normalize these values to create an average)

Since tracking a large bandwidth can take a long time and with many measurement points which all have to have a relative value stored in memory. Here for most spectrum analyzer have a variable called_ bin size/ interval_ which will specify the interval in hertz in which measurements are stored and taken.

If anything is unclear please say so, Love to hear what you think.
Dantali0n~

TonMachielsen, 2016-07-26:
I use rx_power (a Soapy-based rtl_power clone) for this. As mentioned before, scanning a large bandwidth takes a long time and i think CubicSDR is meant to be a real time SDR. Having said that, if Cubic could read rx_power/rtl_power's csv files, that would be awesome.