cjcliffe/CubicSDR #171: Scanner Mode

cjcliffe, 2015-10-30:
Now that there's some leftover horsepower from the PFBCH for background demodulation; it would be handy to add a scanner mode.  Scanner mode could run a demodulator across the band at relevant locations and tag signal peaks and bandwidths.  A second pass could then spend more time and rotate the demodulation modes (analog and digital) in an attempt to figure out what the signal is.  Once the signal has been identified it can be displayed visually and used with some sort of activation marker or gesture and added to the transient bookmark/history list in #161 so that it can be quickly saved or used.

- [ ] Add a key/button to toggle continuous scanner mode
- [ ] Display signals visually and log them to #161
- [ ] Allow click of marker or log entry to initiate demodulation
- [ ] Add an option for "deep-scanning" mode that scans modulation types
- [ ] Allow one-shot usage of deep-scanning when manually placing demodulators

AlmightyOatmeal, 2016-11-18:
This feature is also something that I would absolutely love!

ziermmar, 2018-02-20:
Call me interested in that!

a-raccoon, 2018-04-01:
Question: What are the PFBCH and background demodulation, in contrast to a normal demodulator?  I assume it has to do with how signals are detected and displayed on the waterfall with different color intensity.  Do these generic RTL processes have special jargon names, or CubicSDR function names?

I'm not sure how to ask, but I'm curious if there's a way to use "cheap squelch" to assign a demodulator to a frequency without it having to constantly demodulate that frequency when there is no perceived signal.  In this way, slower computers can "bookmark" more frequencies to be monitored without CubicSDR slowing to a crawl.  A pseudo-scanner mode through inexpensive squelching.

cjcliffe, 2018-05-08:
@a-raccoon PFBCH is the Polyphase Filter Bank CHannelizer -- which efficiently breaks up the wide-band input stream from SDR device into much smaller channels for delivering to the demodulators.  It's primary use is actually for orthogonal frequency-division multiplexing (OFDM) which is a modulation scheme that transmits a signal across multiple carrier frequencies.

The original sentence might have been confusing as it's only freeing up resources which could be re-allocated to some background demodulation and scanning of the channels to evaluate possible signals.

I do like the "Cheap Squelch" idea and have considered it -- but it would require running additional FFTs as the ones currently displayed are optimized in a purely forward manner for display and aren't much use for DSP feedback (i.e. zooming in makes non-visible parts of the FFT unusable).   

I think improving the current squelch should be more of a priority than introducing alternatives in any case.

a-raccoon, 2019-04-21:
:'(

christiaanbrand, 2020-10-28:
:(