cjcliffe/CubicSDR #173: OpenGL 3.x update

cjcliffe, 2015-10-31:
Import and utilize/improve the new shader system from new CubicVR2/CubicFX codebase and port current visuals.

- [ ] Update legacy/prototype OpenGL 1.x code
- [ ] Import CubicVR2 base code for OpenGL 3.x
- [ ] Create relevant 1:1 replacement shaders for visuals
- [ ] Create new shaders to shift CPU graphics and prep work to more suitable GPU shader functions
- [ ] Contemplate the magnitude of possibilities for new visuals :)

cjcliffe, 2016-03-21:
wxWidgets 3.1.0 has been released at https://www.wxwidgets.org/ most notably:
- Revamped OpenGL support better suited to modern OpenGL (3.2+).
- Better support for high DPI displays, especially under Windows.
- OpenGL ES support -- This should make it possible to run properly on Raspberry PI!

Coder206, 2017-04-01:
@cjcliffe Looking forward for the high DPI display support on my Windows machine! Keep up the great work! :)