cjcliffe/CubicSDR #178: [Feature Request] Peak Hold

tomswartz07, 2015-11-05:
Hi there!

First off, thanks so much for this project.
I'm amazed by how many great features are available in this program, even compared to other SDR software that's developed by huge teams. :+1: 

I would love to have a 'Peak Hold' feature built in to the program.
To be precise, this would be a thin line that highlights the highest dB level at each point in the spectrum graph.

Would it be possible to have this implemented?

cjcliffe, 2015-11-08:
Thanks! I'm glad it's working well for you.

Sounds like this might be a useful addition; do you have a screenshot or illustration of what you're describing?

vsboost, 2015-11-08:
Hey Charles,

Something like this i believe tomswartz07  is talking about.  Handy to see where a transmission has taken place if your not looking at the screen.

[image]

tomswartz07, 2015-11-08:
That's exactly the thought, vsboost.

Thanks!

cjcliffe, 2015-11-11:
Thanks, this certainly looks doable -- I initially thought you were referring to the peak-finding dots I see in some screenshots.

I'll add a checklist to the issue and queue it up.

cjcliffe, 2016-01-01:
Peak hold feature is now committed to master; just click the green 'P' to activate/deactivate.

Cheers!

Toontje, 2016-01-01:
Can you remove the green line whenever a different frequency is chosen. Now, when i turn the VFO of the radio i get multiple peaks for he same signal.
If you look at the below image you see only 4 peaks on the white line and multiples on the green line. This is because i turned the VFO on the radio which makes the signal wonder over the band.

[image]

vsboost, 2016-01-01:
> ![screen shot 2016-01-01 at 20 52

The green line should reset it self when you move frequency, well it does for me, I change frequency using the center frequency box.

cjcliffe, 2016-01-02:
@Toontje hmm, yeah it should probably reset under more conditions -- I've attempted to make it track the peak but there's a bit of drift at the moment.   I'll do some more testing and tweaking over the next couple days.

cjcliffe, 2016-01-02:
@Toontje I've played with it some more, and trying to keep/merge and instantly restart peak capture data as soon as possible was causing a lot of usability problems.

I've added a 30 frame delay between peak hold resets; and movement will reset the peak instead of trying to drag it around (which caused aliases and other oddities to emerge).   Right-clicking spectrum also lets you perform a peak-hold reset without having to toggle the peak hold button on and off.

Let me know what you think.

vsboost, 2016-01-02:
Working well here.

Toontje, 2016-01-02:
I understand that you didn't understand my use case. ;-)
In the video link attached you can see what i mean. First part of the video is the behaviour you intended to create. I move the spectrum in CubicSDR and the Peak Hold resets.
Second part of the video is my use case where i don't touch CubicSDR but change the frequency on the VFO of the radio. This is where the Peak Hold doesn't reset. 
Now that i am writing this i realise that what i want to see doesn't make any sense at all. You cannot reset the Peak Hold based on the signal you receive. Maybe later when CAT support is enabled you can reset the Peak Hold whenever CubicSDR gets the information from the radio that the frequency has changed.
Video here: https://www.dropbox.com/s/yqxwzcgsa3e1byj/PeakHold.mov?dl=0

cjcliffe, 2016-01-02:
@Toontje ah yes, we're going to require the CAT control for that.  I've been tinkering with hamlib and figuring out where to start fitting it into SoapyAudio for basic I/Q radio control.  I'll likely end up adding direct support for hamlib within CubicSDR for some advanced features as well once it's working for the basic radio control case.  

I'll get something up shortly so you can let me know how it goes; I think I'll just be able to emulate with an Arduino or a serial cable and some other software here at the moment.

Toontje, 2016-01-02:
I can only help you with the Kenwood protocol, but i do know a lot of radio people who i can convince testing CubicSDR with other brand radios. No worries, we'll figure something out.

cjcliffe, 2016-01-02:
Ok, kenwood protocol looks pretty simple -- https://github.com/N0NB/hamlib/tree/master/kenwood -- I'll get my implementation pushed to SoapyAudio and try a quick emulation of it here on an Arduino as well.

cjcliffe, 2016-01-03:
@Toontje going to close this one up now and move conversation over to #70

closed 2016-01-03 by cjcliffe