cjcliffe/CubicSDR #179: Improvements to CubicSDR

frspin, 2015-11-05:
For this already  good SDR program this improvements can be useful in HF usage:
- an audio bandwidth change changing decoding mode as:
  FM as now bw=200000 centered
  AM BW=6000 centered
  USB BW=200-2700 upper
  LSB BW=200-2700 lower

- a NBFM mode
- an audio spectrum with a graphical envelope for BW, with mouse interaction for modify BW
- some DSP on audio as NB, NR, Notch (auto or manual)
- CAT control as client, for using CubicSDR with Fldigi/WSJT-X/etc.

Regards
Franco Spinelli
IW2DHW

frspin, 2015-11-07:
A very raw code for first point, related to changing bandwidth when changing demodulator type, can be in AppFrame.cpp:

            if (dSelection != -1 && dSelection != demod->getDemodulatorType()) {
                demod->setDemodulatorType(dSelection);

                // start addition
                demodBw = 200000;
                if (dSelection > 2) {
                    demodBw = 3000;
                }
                if (dSelection == 2 || dSelection == 5) {
                    demodBw = 6000;
                }
                demod->setBandwidth(demodBw);
               // end addition
            }
            int muteMode = demodMuteButton->getSelection();
            if (demodMuteButton->modeChanged()) {

Franco Spinelli
IW2DHW

cjcliffe, 2015-11-08:
Thanks Franco, I'll likely break this up into separate issues as there's a fair bit of enhancements.

I'll comment at least for now on the bandwidth defaults;  I think that makes perfect sense as I find myself manually entering the demodulator bandwidth to those same defaults when I change types regularly.

I think what I'll do for that is set those as the defaults; and in each case when you change the bandwidth it will remember the last bandwidth for that particular demodulation type.  So if you toggle it back and forth or create more you won't lose any manual adjustments.

I think NBFM could be achieved by clicking the FM button again to possibly toggle between Wide (with default stereo pilot detect) and narrow.  It wouldn't be limited to just NBFM though it'd just be a secondary bandwidth setting for the FM you could just toggle back and forth and set separately.

Will put in some more thoughts later, thanks!

frspin, 2015-11-08:
Il 08/11/2015 05:15, Charles J. Cliffe ha scritto:
> Thanks Franco, I'll likely break this up into separate issues as there's
> a fair bit of enhancements.

Feel free to break this in as many issues as needed.

> I'll comment at least for now on the bandwidth defaults; I think that
> makes perfect sense as I find myself manually entering the demodulator
> bandwidth to those same defaults when I change types [...]

Good solution. It is not so friendly to give a double bandwidth for USB
and LSB, due to demodulator solution using resample2_crcf_filter_execute().
I think it is better to give a bandwidth of 2700 Hz and internally use a
bandwidth of 5400 Hz.

> 
> I think NBFM could be achieved by clicking the FM button again to
> possibly toggle between Wide (with default stereo pilot detect) and
> narrow. It wouldn't be limited to just NBFM though it'd just be a
> secondary bandwidth setting for the FM you could just toggle back and
> forth and set separately.

Also this is a good solution. You can change button label according to
FM choice. And also use a no-deemphasis demodulation for NBFM

Regards
Franco Spinelli
IW2DHW

frspin, 2015-11-11:
There is some problem in bandwidth use with USB or LSB.
The current demodulation is due on samples in full bandwidth, removing unwanted band (USB or LSB) with a filter and then demodulating signal.
With this solution bandwidth is double of required. If you reduce bandwidth, this reduction is centred on tune frequency. This modify the decoded pitch of demodulated signal.

As you can see in attached files, red line (white in a fax signal) is moving in bandwidth changing this position from 6000 to 5000. This in wrong. 
[image]
[image].

May be another demodulation scheme is required?

Regards
Franco Spinelli
IW2DHW

cjcliffe, 2015-11-11:
@frspin I believe it shouldn't be much problem to make USB/LSB show the correct bandwidth -- I'll create a branch soon with the changes and have you check it out to let me know if it's any better.    

The USB/LSB is technically just a non-carrier mode of AM in liquid-dsp; but I'm not sure that it's completely ignoring the 'carrier' in SSB mode as I've had many issues with the frequency drifting above certain bandwidths.  I will check with @jgaddert and review the relevant liquid-dsp code to make sure it is in fact ignoring carriers and not performing any shift in SSB mode.

jgaeddert, 2015-11-12:
How are you invoking the ampmodem demodulator? It's possible I have a typo in the code.

cjcliffe, 2015-11-12:
@jgaeddert thanks for popping by :)  The relevant code for ampmodem init is:

```
    demodAM = ampmodem_create(0.5, 0.0, LIQUID_AMPMODEM_DSB, 0);
    demodAM_USB = ampmodem_create(0.5, 0.0, LIQUID_AMPMODEM_USB, 1);
    demodAM_LSB = ampmodem_create(0.5, 0.0, LIQUID_AMPMODEM_LSB, 1);
    demodAM_DSB = ampmodem_create(0.5, 0.0, LIQUID_AMPMODEM_DSB, 1);
```

I'm assuming the 'carrier suppressed' means 'don't detect carrier' but I still hear it shifting pitch on the audio output when there's a steady tone or heavy amount of CW at higher bandwidths.

cjcliffe, 2015-11-12:
@jgaeddert Just looking at the liquid-dsp code it seems correct to me, it only performs the mixing in the case when the carrier is present..

closed 2015-12-08 by cjcliffe