cjcliffe/CubicSDR #184: Error building CubicSDR on OSX 10.10

Toontje, 2015-11-10:
```
Checking connectivity... done.
-- The C compiler identification is AppleClang 7.0.0.7000176
-- The CXX compiler identification is AppleClang 7.0.0.7000176
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
64 bit compiler detected
-- Found OpenGL: /System/Library/Frameworks/OpenGL.framework  
-- Found FFTW: /opt/local/lib/libfftw3.dylib  
-- Found LIQUID: /opt/local/lib/libliquid.dylib  
-- Found wxWidgets: TRUE  
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/ton/CubicSDR/build
CPack: Create package using DragNDrop
CPack: Install projects
CPack: - Run preinstall target for: CubicSDR
CPack: - Install project: CubicSDR
  exe_dotapp_dir/='/Users/ton/CubicSDR/build/x64/CubicSDR.app/'
  item_substring='/opt/local/Library/Frameworks/wxWidgets.fra'
  resolved_embedded_item='/opt/local/Library/Frameworks/wxWidgets.framework/Versions/wxWidgets/3.0-libcxx/lib/libwx_baseu-3.0.0.2.0.dylib'

Install or copy the item into the bundle before calling fixup_bundle.
Or maybe there's a typo or incorrect path in one of the args to fixup_bundle?

CMake Error at /opt/local/share/cmake-3.3/Modules/BundleUtilities.cmake:737 (message):
  cannot fixup an item that is not in the bundle...
Call Stack (most recent call first):
  /opt/local/share/cmake-3.3/Modules/BundleUtilities.cmake:848 (fixup_bundle_item)
  /Users/ton/CubicSDR/build/cmake_install.cmake:34 (fixup_bundle)

CPack Error: Error when generating package: CubicSDR
```

cjcliffe, 2015-11-10:
Unfortunately you can't bundle the framework version of wxWidgets properly.  Check the new build instructions for how to build the static version of wxWidgets and make sure your wx-config is pointed at the built version.

Toontje, 2015-11-11:
I had the path to the wxWidgets wrong. Sorry for creating the ticket. I updated the build docs on the wiki.

closed 2015-11-11 by Toontje