cjcliffe/CubicSDR #185: Linux Packaging -- what's the best options here?

cjcliffe, 2015-11-10:
Looking for some help and to start some discussion on how to best package CubicSDR for deployment on Linux.  I'm fairly inexperienced with creating packages and the CMake .deb builder is about as far as I've gotten to date.

Looking for some ideas and assistance on where this should go next.

k9wkj, 2015-11-11:
Deb !  of course i say Deb  as im a old debian fart

the Arch folks will bend over backwards to help out for AUR
have you ever noticed that the Arch wiki is the best info on the planet?

and im sure the Buntu folks would help with PPA packages

thats all i have for now

tomswartz07, 2015-11-11:
I'd be super interested in providing a PKGBUILD for the Arch AUR.

Unfortunately, 80% of my time is spent on my work LAN where I don't have write access to the AUR git repository, so I can't post the PKGBUILD as easily as I would like. :cry:

tomswartz07, 2015-11-11:
Hi there:

I had some time at lunch today, so I whipped up the necessary PKGBUILDS for ArchLinux.
https://github.com/tomswartz07/CubicSDR-PKGBUILDS

I might have some time later tonight to push them to the AUR.
Is this acceptable?

cjcliffe, 2015-11-12:
@k9wkj thanks; I think the DEB is decent except for my dependency definitions; I'll have to figure out what it needs for supporting more distros.

@tomswartz07 excellent thanks; I can definitely add a "make install" that installs it to the .DEB locations.

cjcliffe, 2015-11-14:
@tomswartz07 I've added a 'make install' target as well as updated the CubicSDR.desktop to respect the CMAKE_INSTALL_PREFIX.  There's also a 'make uninstall' target now as well.

guruofquality, 2015-11-14:
**Ubuntu**
For Pothos, SoapySDR, and support modules, I have been versioning a debian/ directory along with the source. And using git-buildpackage, I have been creating source packages which get uploaded to a PPA and built for several arches: https://launchpad.net/~myriadrf/+archive/ubuntu/drivers

Making a debian/ directory is more tedious than the CPack DEB, but its a good way to get into PPA, and also a way to get proper multi-package builds. I think I can help out here with that.

**OpenSuse**
This user maintains OpenSuse builds for a variety of SDR related software packages, including soapy sdr and friends. So that's been pretty awesome, and I think he would be kind enough to host one more: https://build.opensuse.org/project/show/home:mnhauke:sdr

**PyBombs**
Also, there is a source-based build system used within this SDR niche called PyBombs. Its very similar to homebrew (but for linux) and about the same amount of work to support. I have an open issue here: https://github.com/pothosware/SoapySDR/issues/51

**Others?**
Anyway, packaging for linux is very broad and diverse -- but I would like to try and support whatever else comes out of this discussion. Looks like creating a repo for arch linux PKGBUILD build scripts may be a worthwhile effort as well.

tomswartz07, 2015-11-18:
In regards to the ArchLinux AUR scripts: It's easy to build a package from any source.

I'm pretty confident that no matter what packaging solution is chosen, we could create a PKGBUILD for Arch easily.
For example: there's a [PKGBUILD for Google Chrome](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=google-chrome) that uses the .DEB files directly from Google as the source.
My point being: don't let Arch dictate how you package the software. One of us will figure out a way to make it work. :smile: 

From my perspective, as referenced in #190, it's becoming a bit difficult to stay on top of the different dependencies- "Dependency Hell", as I've heard it called.
I think this should be the top priority. Once it's figured out how to easily tie in all of the dependencies, then it's easy to package it up with whatever service you choose (.deb, Suse build system, etc).

If I could offer a bit of perspective from a few other projects:

1. ownCloud uses [openSuse Build Service](https://build.opensuse.org/search)
2. GQRX seems to have the [dependencies listed in the CMAKE file](https://github.com/csete/gqrx#installation-from-source), but that project also appears to intelligently pick between the SDR backends that are installed on the client computer at build time.

Perhaps a mish-mash of those two are used?
If the user already has RTLSDR stuff installed (i.e. pre-SoapySDR changes), then use that. If they have SoapySDR or other devieces, use that instead?

cjcliffe, 2015-11-18:
Thanks for the help so far everyone, looking like several solutions/targets here I wasn't aware of before.  

@tomswartz07 did that 'make install' target do what you needed it to do?

Not sure if I'd go for the submodule route right away but I might be convinced -- it's just caused problems for me in the past.   Including a cmake target or shell script to check-out and build SoapySDR and a set of BOOL flags for other modules to build might be a start. 

Personally I just have a .bat/.sh file for each configuration (release/debug / 64/32) that I've been using here to update, build and install everything in one go and CubicSDR just bundles whatever is in the soapy paths during installer build.  Those could be modified and included as a 'bootstrap' script with a few options to download / build soapy module dependencies for the platform.

stefantalpalaru, 2015-12-08:
I made Gentoo packages which might be useful as a starting point for other packagers: https://github.com/stefantalpalaru/gentoo-overlay/tree/master/net-wireless

This patch is relevant for upstream (needed with wxGTK-3.0.2.0): https://github.com/stefantalpalaru/gentoo-overlay/blob/master/net-wireless/cubicsdr/files/cubicsdr-0.1.16-wxwidgets.patch

This one too: https://github.com/stefantalpalaru/gentoo-overlay/blob/master/net-wireless/liquid-dsp/files/liquid-dsp-9999-libdir.patch

viraptor, 2016-01-26:
If you can make fpm (https://github.com/jordansissel/fpm) work with cmake, that may be a good solution for you. It basically tries to abstract the system differences, so once you get it working, you can get debs, rpms and whatever else with little-to-no modification.

cjcliffe, 2016-01-26:
@stefantalpalaru thanks -- I'll try and get all the known packages listed somewhere; I think someone's already done an Arch one as well.

@viraptor I'll definitely take a look at FPM -- seems like a good way to get multiple packages out of one tool; thanks.

viraptor, 2016-01-26:
Yes, arch already works. https://aur.archlinux.org/packages/cubicsdr-git/

cjcliffe, 2016-02-28:
This seems like a good way to produce an all-inclusive linux package, any thoughts?

https://github.com/probonopd/AppImageKit

cjcliffe, 2016-03-16:
I've added an experimental AppImageKit Linux build to https://github.com/cjcliffe/CubicSDR/releases/tag/0.1.26-alpha-experiment and I'd be happy to hear how/if it works out for anyone.

Cheers!

vsboost, 2016-03-16:
Hi Charles,

Downloaded the file and set +x then ran.

14:58:51: Warning: Mismatch between the program and library build versions detected.
The library used 3.0 (wchar_t,compiler with C++ ABI 1009,wx containers,compatible with 2.8),
and your program used 3.0 (wchar_t,compiler with C++ ABI 1002,wx containers,compatible with 2.8).
Segmentation fault

cjcliffe, 2016-03-16:
@vsboost thanks, I tried the first build with an old ubuntu VM to see how far back I could go, I'll try a VM with newer C++ ABI and add a test build.

cjcliffe, 2016-03-17:
@vsboost I've added a new build based on ubuntu 15 and the new automated buildAppImage.sh script; let me know if that works any better.  The script should make it easy to build several bundles that match various distributions I'll just need to figure them all out so there's a minimal set of binaries :)

vsboost, 2016-03-17:
Hi Charles,

How do i use this?

Sorry for the silly question.

cjcliffe, 2016-03-17:
@vsboost sorry I wasn't clear on that; just try the latest "CubicSDR" file build on https://github.com/cjcliffe/CubicSDR/releases/tag/0.1.26-alpha-experiment that's based on ubuntu 15. 

Note that I renamed the old one you tried before to CubicSDR-ubuntu14.

vsboost, 2016-03-18:
All good, tested the binary on Kali Rolling and works, can i just ask one favor? are you able when you get a chance to bundle SoapyHackRF.

Cheers

piotr-szegda, 2016-03-18:
I can confirm it's work great on Ubuntu 16.04 but you should mention in the documentation that you need install first rtl-sdr and blacklist dvb_usb_rtl28xxu blacklist rtl2832 blacklist rtl2830.
How to do that:
apt install rtl-sdr
echo "blacklist dvb_usb_rtl28xxu blacklist dvb_usb_v2 blacklist rtl_2830 blacklist rtl_2832 blacklist r820t" >> /etc/modprobe.d/blacklist.conf

cjcliffe, 2016-03-18:
@vsboost HackRF and some others were omitted to keep the build simple, now that I've got some confirmation of it working I'll add more for the next test.

@piotr-szegda I can understand the blacklisting but the rtl-sdr support should have been included -- can you confirm that it required doing both or does it still work if you remove the rtl-sdr system package and just blacklist the module?

piotr-szegda, 2016-03-18:
@cjcliffe: I have permission problem without rtl-sdr. Udev rules should by OK too but rtl-sdr install is much simpler.

 `
usb_open error -3
Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules
  origin=https://github.com/pothosware/SoapyRTLSDR
  rtl=0
`

piotr-szegda, 2016-03-18:
@cjcliffe:
Now I check and librtlsdr0 which is a dependency of rtl-sdr have udev rules and necessary blacklist.
So we can install librtlsdr0 instead of rtl-sdr.
http://packages.ubuntu.com/pl/xenial/amd64/librtlsdr0/filelist

cjcliffe, 2016-03-19:
@vsboost I've added HackRF and BladeRF to the latest Linux build at https://github.com/cjcliffe/CubicSDR/releases/tag/0.1.26-alpha-experiment now packaged as a .tgz so it should already be executable when extracted.

I've removed the original test builds so I'll leave them here for reference/comparison:
[CubicSDR.tar.gz](https://github.com/cjcliffe/CubicSDR/files/180635/CubicSDR.tar.gz)
[CubicSDR-ubuntu14.tar.gz](https://github.com/cjcliffe/CubicSDR/files/180634/CubicSDR-ubuntu14.tar.gz)

vsboost, 2016-03-19:
Thanks Charles, working well so far.

vsboost, 2016-03-19:
Just noticed something, when manually adjusting gains on HackRF the slider bars don't update quickly, it takes a few sec before the bars show the place you clicked.

cjcliffe, 2016-03-19:
@vsboost that sounds.. odd.. I'll give it a try here soon and see if I can reproduce the issue :)

Glad it's working for you; I'm going to spin up a few more VMs and try some more distributions here to see how it goes.

vsboost, 2016-03-19:
Yea i thought it was strange too, i did test a the latest master that i built on the same machine and that one was fine.

closed 2016-08-14 by cjcliffe