cjcliffe/CubicSDR #186: Error when starting CubicSDR

Toontje, 2015-11-11:
No RTL-SDR adapter connected.

```
SoapySDR init..
	API Version: v0.4.0-g4331e634
	ABI Version: v0.4-2
	Install root: /usr/local
	Loading modules... 
[ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules/libremoteSupport.so)
  duplicate entry for null ()
	Available factories...null, remote, rtlsdr, sdrplay
```

cjcliffe, 2015-11-12:
Hmm, I think @guruofquality had something in about this; null module gets included in SoapyRemote as well or something -- it's harmless really since it's not used that's just saying it refused to load it twice..

guruofquality, 2015-11-12:
Explanation and issue for reference: https://github.com/pothosware/SoapySDR/issues/53

cjcliffe, 2015-11-13:
@guruofquality thanks, that's the one -- going to close this one so we're not duplicating.

closed 2015-11-13 by cjcliffe