cjcliffe/CubicSDR #187: stupid user question

k9wkj, 2015-11-11:
i must be a dullard
i have no idea how to control the LNA gain in the dongle
i know there is some agc action going on.
but is there? or could there be a control for auto/manual , gain , attack speed
most other programs have a gain control, GQRX is the only one that gives any control over it

cjcliffe, 2015-11-12:
Is this RTLSDR? and which tuner does the dongle have?

k9wkj, 2015-11-12:
Ohh yes RTL-SDR its a T2 tuner
Well tuners, I have 5 of them

cjcliffe, 2015-11-12:
@k9wkj are you just looking to enable the manual gain? If you have the latest build you can just un-check "Automatic Gain" under the settings menu and the manual gain controls will appear.

k9wkj, 2015-11-12:
ohh my!!!
i havent built latest yet
and i guess i didnt read the change logs!

thanks!

cjcliffe, 2015-11-12:
Aye, been quite a few things pushed to master in the last few days :)

https://github.com/cjcliffe/CubicSDR/releases/tag/0.1.16-soapysdr also has a bundled binary version for OSX and Windows.  

Cheers!

closed 2015-11-12 by cjcliffe