cjcliffe/CubicSDR #189: Sound card based SDRs

pwarren, 2015-11-17:
Hi all, 

Would it be possible to use soundcard based SDRs, such as the Peaberry SDR: 
http://ae9rb.com/index.php?main_page=product_info&products_id=8, 
or Softrock: 
http://fivedash.com/index.php?main_page=product_info&products_id=7 
style radios with CubicSDR?

I'm guessing another module for SoapySDR will be needed?

cjcliffe, 2015-11-17:
@pwarren I've already started the idea over at https://github.com/pothosware/SoapyAudio for just this purpose; please feel free to add these links and any relevant issues there :+1:

cjcliffe, 2015-11-18:
@pwarren thanks I see you've added stuff over at the SoapyAudio repository.  I'll close this issue for now and comment there when the time comes.

closed 2015-11-18 by cjcliffe